Xilinx/XRT #6364: `[xclbinutil]:ps-kernel` test case fails

j-stephan, 2022-03-04:
Using the standard `build.sh` script to compile XRT the `[xclbinutil]:ps-kernel` fails during the `Debug` build:

```
13/14 Test #13: [xclbinutil]:ps-kernel ............***Failed    0.26 sec
>>> Start of unit test <<<
Autocomplete enabled for the xbutil command
Autocomplete enabled for the xbmgmt command
XILINX_XRT        : /home/jan/build/XRT/build/Debug/opt/xilinx/xrt
PATH              : /home/jan/build/XRT/build/Debug/opt/xilinx/xrt/bin:/home/jan/software/scripts:/usr/local/cuda/bin:/home/jan/.cargo/bin:/home/jan/bin:/home/jan/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
LD_LIBRARY_PATH   : /home/jan/build/XRT/build/Debug/opt/xilinx/xrt/lib:
PYTHONPATH        : /home/jan/build/XRT/build/Debug/opt/xilinx/xrt/python:
Starting test
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
0) Create working xclbin container
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
xclbinutil --add-section EMBEDDED_METADATA:RAW:/home/jan/build/XRT/src/runtime_src/tools/xclbinutil/unittests/PSKernel/embedded_metadata.xml --add-section MEM_TOPOLOGY:JSON:/home/jan/build/XRT/src/runtime_src/tools/xclbinutil/unittests/PSKernel/mem_topology.json --add-section IP_LAYOUT:JSON:/home/jan/build/XRT/src/runtime_src/tools/xclbinutil/unittests/PSKernel/ip_layout.json --add-section CONNECTIVITY:JSON:/home/jan/build/XRT/src/runtime_src/tools/xclbinutil/unittests/PSKernel/connectivity.json --output working.xclbin --force
XRT Build Version: 2.13.0 (master)
       Build Date: 2022-03-04 11:34:07
          Hash ID: 4ab537cd66ec21a14e8277d22c286f83dc9b325c
Creating a default 'in-memory' xclbin image.

Section: 'EMBEDDED_METADATA'(2) was successfully added.
Size   : 4424 bytes
Format : RAW
File   : '/home/jan/build/XRT/src/runtime_src/tools/xclbinutil/unittests/PSKernel/embedded_metadata.xml'

Section: 'MEM_TOPOLOGY'(6) was successfully added.
Size   : 88 bytes
Format : JSON
File   : '/home/jan/build/XRT/src/runtime_src/tools/xclbinutil/unittests/PSKernel/mem_topology.json'

Section: 'IP_LAYOUT'(8) was successfully added.
Size   : 88 bytes
Format : JSON
File   : '/home/jan/build/XRT/src/runtime_src/tools/xclbinutil/unittests/PSKernel/ip_layout.json'

Section: 'CONNECTIVITY'(7) was successfully added.
Size   : 40 bytes
Format : JSON
File   : '/home/jan/build/XRT/src/runtime_src/tools/xclbinutil/unittests/PSKernel/connectivity.json'
Successfully wrote (8310 bytes) to the output file: working.xclbin
Leaving xclbinutil.

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2a) Read in a PS kernel, updated and validate the sections
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
xclbinutil --input working.xclbin --add-pskernel pskernel.so --dump-section EMBEDDED_METADATA:RAW:embedded_metadata_updated.xml --dump-section IP_LAYOUT:JSON:ip_layout_updated.json --dump-section CONNECTIVITY:JSON:connectivity_updated.json --dump-section MEM_TOPOLOGY:JSON:mem_topology_updated.json --output pskernel_output.xclbin --force
XRT Build Version: 2.13.0 (master)
       Build Date: 2022-03-04 11:34:07
          Hash ID: 4ab537cd66ec21a14e8277d22c286f83dc9b325c
Reading xclbin file into memory.  File: working.xclbin

ERROR: No cache value found for: '(ref4) <0xcbd>, xrtHandles'

Exception('Operation failed with the return code: 255')
Test Status: FAILED
>>> End of unit test <<<
 
========================================================================
Test Directory    : /home/jan/build/XRT/build/Debug/unit-tests/xclbinutil/ps-kernel
Log File          : runme.log
Error Return Code : 1
Test Status       : FAILED
========================================================================
```

I'm working on Ubuntu 21.10, kernel 5.13 and compiled everything with GCC 11.2.

j-stephan, 2022-03-04:
The offending PR is #6297. Reverting to the commit immediately before (`05e9421353acc9d3791e06b0ac7e8fda150b3cd8`) works.

uday610, 2022-03-30:
After reviewing this internally we believe this issue is fixed. So are closing this. Thank you

closed 2022-03-30 by uday610

keryell, 2022-04-12:
Yes, this is solved for me too. Thanks.