Xilinx/XRT #6382: Xrt can not correctly recognize the clock freq. in v2021.2

doonny, 2022-03-09:
I am using vitis version 2021.2 with the zcu102_base_dfx platform.   In the generated summary file, the clock frequency was not correctly recognized from the hardware, for example, the default clock is 150Mhz(id=0) in the platform, but when I set the clock to 300Hhz(id =1), in the summary file, the clock was still 150Mhz:
[image]

So the utilization became 195.8%>100% .  And other data, like bandwidth, were also not correctly calculated.

jvillarre, 2022-10-19:
This is a known problem with clocks set to some frequency other than the default.  It is currently in the plan to fix in the first half of 2023.

doonny, 2023-12-27:
@jvillarre Hello, did v2023.1 fix this issue ?  Is there any way that we could mannually fix it in previous versions like v2022.1 ?