Xilinx/XRT #6383: Tests script board.sh out of date with xbutil

HFTrader, 2022-03-09:
XRT version: 1.23.0  e98799314cbb15e3b174ab0f8b7ba66587d0412e
Alveo u50 xilinx_u50_gen3x16_xdma_201920_3  F465B0A3-AE8C-64F6-19BC-150384ACE69B
[xbutil_query.txt](https://github.com/Xilinx/XRT/files/8214818/xbutil_query.txt)
[xbmgmt_scan.txt](https://github.com/Xilinx/XRT/files/8214819/xbmgmt_scan.txt)
[dmesg.txt](https://github.com/Xilinx/XRT/files/8214820/dmesg.txt)

I have compiled XRT/master and following the [test procedure](https://xilinx.github.io/XRT/master/html/test.html) I attempted to run the test script
```bash
mkdir tests
cd tests
<path>/XRT/build/board.sh -board u50 -sync
```
However it seems that the script expects the old version of the tools (legacy). For once, it calls `xbutil scan` and greps for the result to find the device number. 
```bash
...
export DSA=`${XILINX_XRT}/bin/xbutil scan | grep '\[0\]' | cut -d':' -f5 | cut -d'(' -f1`
```
I tried adding the `--legacy` option after `xbutil scan` but then the DSA comes empty as the grepping also seems to be off. 
I replaced the line with just `export DSA=0000:4b:00.1` but other problems occured and I gave up.

stsoe, 2022-03-16:
board.sh is not supported or maintained.

HFTrader, 2022-03-17:
Soren, understood.
But this script is explicitly mentioned in the documentation - in master!

https://xilinx.github.io/XRT/master/html/test.html

stsoe, 2022-03-17:
> But this script is explicitly mentioned in the documentation - in master!
> 
> https://xilinx.github.io/XRT/master/html/test.html

@HFTrader Thanks for pointing that out, we will have to update the documentation, + @uday610 
I still use the script, just not for harvesting, which has been broken ever since internal test repos where re-organized.
I have not had the time or energy to update the script.

PR #6437 (2022-03-17): Fix wording in test documentation
#### Problem solved by the commit
Fix obsolete references to `board.sh` in documentation.
Closes #6383.  

#### Bug / issue (if any) fixed, which PR introduced the bug, how it was discovered
The board.sh script is not supported, but can be used with manually harvested tests.
commit 2d1e4cb3 (2022-03-17): Fix wording in test documentation (#6437)

Closes #6383.  The board.sh script is not supported, but can be
used with manually harvested tests.
closed 2022-03-17 by stsoe