Xilinx/XRT #6424: Feature Request for build_xrt_with_docker.sh

bryanloz-xilinx, 2022-03-16:
Feature Request for [build_xrt_with_docker.sh](https://github.com/Xilinx/XRT/blob/master/src/runtime_src/tools/scripts/build_xrt_with_docker.sh)

- Add Support for directing build products to a user specified output directory
- Add Support for centos

```
Internally at Xilinx we often face problems building XRT for CentOS, because the Centos YUM repositories are controlled by IT
So it would be nice to have the freedom to build in a clean CentOS inside of Docker.
```

stsoe, 2022-03-16:
I believe @bryanloz-xilinx added a script for this

bryanloz-xilinx, 2022-03-17:
@stsoe I wanted to leave this open to track these incomplete features.
At the moment the script only supports Ubuntu 18.04 and 20.04

stsoe, 2022-03-17:
Open it and assign it to yourself then. It didn't have an owner.