Xilinx/XRT #6444: RHEL 8 (dkms 4.18): error: ‘struct drm_device’ has no member named ‘pdev’;

alexisfrjp, 2022-03-18:
```
  CC [M]  /var/lib/dkms/xrt/2.13.0/build/driver/xocl/userpf/../../common/xrt_xclbin.o
  CC [M]  /var/lib/dkms/xrt/2.13.0/build/driver/xocl/userpf/xocl_drv.o
  CC [M]  /var/lib/dkms/xrt/2.13.0/build/driver/xocl/userpf/xocl_errors.o
  CC [M]  /var/lib/dkms/xrt/2.13.0/build/driver/xocl/userpf/xocl_bo.o
  CC [M]  /var/lib/dkms/xrt/2.13.0/build/driver/xocl/userpf/xocl_drm.o
/var/lib/dkms/xrt/2.13.0/build/driver/xocl/userpf/xocl_drm.c: In function ‘xocl_drm_init’:
/var/lib/dkms/xrt/2.13.0/build/driver/xocl/userpf/xocl_drm.c:576:8: error: ‘struct drm_device’ has no member named ‘pdev’; did you mean ‘dev’?
  ddev->pdev = XDEV(xdev_hdl)->pdev;
        ^~~~
        dev
make[3]: *** [scripts/Makefile.build:316: /var/lib/dkms/xrt/2.13.0/build/driver/xocl/userpf/xocl_drm.o] Error 1
make[2]: *** [Makefile:1577: _module_/var/lib/dkms/xrt/2.13.0/build/driver/xocl/userpf] Error 2
make[2]: Leaving directory '/usr/src/kernels/4.18.0-365.el8.x86_64'
make[1]: *** [Makefile:133: all] Error 2
make[1]: Leaving directory '/var/lib/dkms/xrt/2.13.0/build/driver/xocl/userpf'
make: *** [Makefile:2: all] Error 2
```

Kernel version: **4.18.0-365.el8.x86_64**

alexisfrjp, 2022-03-18:
Here is the patch:

```
diff --git a/src/runtime_src/core/pcie/driver/linux/xocl/userpf/xocl_drm.c b/src/runtime_src/core/pcie/driver/linux/xocl/userpf/xocl_drm.c
index 17ce660a8..99db738b8 100644
--- a/src/runtime_src/core/pcie/driver/linux/xocl/userpf/xocl_drm.c
+++ b/src/runtime_src/core/pcie/driver/linux/xocl/userpf/xocl_drm.c
@@ -571,9 +571,16 @@ void *xocl_drm_init(xdev_handle_t xdev_hdl)
 	 * The pdev field was removed from drm_device starting from 5.14 and
 	 * should be skipped starting from that version.
 	 * https://github.com/torvalds/linux/commit/b347e04452ff6382ace8fba9c81f5bcb63be17a6
+	 * Also from RHEL 8.6
 	 */
+#if defined(RHEL_RELEASE_VERSION)
+#if RHEL_RELEASE_CODE < RHEL_RELEASE_VERSION(8, 6)
+  ddev->pdev = XDEV(xdev_hdl)->pdev;
+#endif
+#else
 #if LINUX_VERSION_CODE < KERNEL_VERSION(5, 14, 0)
 	ddev->pdev = XDEV(xdev_hdl)->pdev;
+#endif
 #endif
 
 	ret = drm_dev_register(ddev, 0);
```

Tested, working.

uday610, 2022-03-30:
Thanks, you can probably send a PR that we can review for patching. Thanks

keryell, 2022-05-19:
What is the status of this? Did it landed?

rbramand-xilinx, 2022-05-19:
Hi @keryell, I started working on this. The changes will be pushed in 1-2 days

alexisfrjp, 2022-05-19:
> Thanks, you can probably send a PR that we can review for patching. Thanks

I wrote the patch above, you guys just needed to take it and merge it...

alexisfrjp, 2022-05-19:
> What is the status of this? Did it landed?

I wrote the patch above, you guys just needed to take it and merge it...

rbramand-xilinx, 2022-05-20:
@alexisfrjp there are other changes as well, I will send PR including all changes to support RHEL 8.6

rbramand-xilinx, 2022-05-26:
PR- https://github.com/Xilinx/XRT/pull/6734 solves this issue