Xilinx/XRT #6585: clCreateProgramWithBinary and DFX platform

dj-park, 2022-04-08:
I am using ZCU102 DFX platform, and a single ZCU102 board. I used `pr_subdivide` command to create multiple reconfigurable pblocks. Then, I generated multiple partial bit files and xclbin files. Let's say I have A.xclbin, B.xclbin, and C.xclbin.

What I am doing is `cl::program` the device with A.xclbin and B.xclbin respectively, and create kernel with `cl::Kernel krnl_test(program, "kernel_test", &err)`. Note that `program` is associated with B.xclbin. Then, I tried to `cl::Program` with C.xclbin, and I got an error of 
`"[XRT] ERROR: device '0000:00:00.0' programmed with different xclbin"`

Below is the part of clCreateProgramWithBinary.cpp, and it compares uuid associated with the `program` and the current xclbin.
The uuid associated with the `program` must be B.xclbin, and it seems to be assigned when I `cl::Kernel`. The current xclbin is C.xclbin, and that's why it's throwing an error above. Is there any workaround to properly load multiple partial bitstreams?

```
    // Compare program uuid against this binary, must match
    auto uuid = program->get_xclbin_uuid(device);
    auto binary = binaries[idx];   // guaranteed not nullptr 
    xuid_t xuuid;
    xclbin_uuid(binary, xuuid);

    if (uuid != xuuid)
      throw xocl::error(CL_OUT_OF_RESOURCES,
                        "device '" + device->get_bdf() +"' programmed with different xclbin");
```

Thanks in advance.

stsoe, 2022-04-13:
@dj-park Have you released the previous program (B.xclbin) before programming the device with C.xclbin.  It sounds like there is some OpenCL resource that is not properly released.

dj-park, 2022-04-14:
@stsoe  Thanks for the reply. In fact, what I am trying to achieve is programming the device with C.xclbin while the B.xclbin's program is running. C.xclbin is independent from B.xclbin's program, so  theoretically, I should be able to program the device.

stsoe, 2022-04-14:
@dj-park That is not supported.   You can only have one xclbin programmed / active at any given time.

closed 2022-04-29 by dj-park

dhjoo98, 2025-03-16:
hello, posting a question 3 years later. Is programming (having active) multiple xclbins at once still not supported in XRT/OpenCL? In vivado, it is possible to specify a partial reconfiguration region and swap in partial bitstreams to reprogram that region. Wondering if the same can be achieved from a host-side toggle.

dhjoo98, 2025-03-17:
@stsoe Thank you for the insight. It's very interesting that partial reconfiguration (i.e. https://docs.amd.com/r/2023.1-English/ug909-vivado-partial-reconfiguration/Introduction) can be done via Vivado interface. Then, the only way for a host + PCIE Alveo card do achieve this in a program, as far as I can tell, is through a tcl script.