Xilinx/XRT #6610: EmbeddedSw aienginev2 include of xaiengine.h is not xaiengine/xaiengine.h

kiwichris, 2022-04-14:
The embeddedsw repo shows the install path for `aienginev2` is `$(INCLUDEDIR)/xaiengine`: 

https://github.com/Xilinx/embeddedsw/blob/master/XilinxProcessorIPLib/drivers/aienginev2/src/Makefile.Linux#L28

The XRT code references the headers directly and not via the include path's install location:
```
$ grep -r 'xaiengine.h' xrt
xrt/src/runtime_src/core/edge/user/aie/aie.h:#include <xaiengine.h>
xrt/src/runtime_src/core/edge/user/aie/common_layer/adf_profiling_api.cpp:#include "xaiengine.h"
xrt/src/runtime_src/core/edge/user/aie/common_layer/adf_runtime_api.cpp:#include "xaiengine.h"
xrt/src/runtime_src/core/edge/user/aie/common_layer/fal_util.h:#include <xaiengine.h>
xrt/src/runtime_src/core/edge/user/aie/graph.cpp:#include <xaiengine.h>
xrt/src/runtime_src/core/edge/user/device_linux.cpp:#include <xaiengine.h>
xrt/src/runtime_src/xdp/profile/plugin/aie_debug/aie_debug_plugin.h:#include <xaiengine.h>
xrt/src/runtime_src/xdp/profile/plugin/aie_profile/aie_plugin.h:#include <xaiengine.h>
xrt/src/runtime_src/xdp/profile/plugin/aie_trace/aie_trace_plugin.h:#include <xaiengine.h>
```
Should the include statements be `#include <xaiengine/xaiengine.h>`?

I am not working with XRT on Linux and so it is not clear to me how the headers are being installed. I do not know if the location embeddedsw installs to is correct or do I need to add a specific `-I` path to the compiler.

larry9523, 2022-04-14:
While there a quite a few AIE driver header files are under 
/usr/include/xaiengine
xaiengine.h is actually under
/usr/include/

https://github.com/Xilinx/embeddedsw/blob/master/XilinxProcessorIPLib/drivers/aienginev2/src/Makefile.Linux#L30

kiwichris, 2022-04-14:
Ah yes it is. I will place that header at the top level in my build.

Thanks for the quick response and sorry for the noise.

closed 2022-04-14 by kiwichris