Xilinx/XRT #6619: why aren't xrt::bo uniqued ?

Ralender, 2022-04-19:
for example:
```
#include <xrt/xrt_kernel.h>
#include <xrt.h>
#include <cassert>

int main() {
auto a = xrt::device(0);
auto b = xrt::device(0);
assert(static_cast<xclDeviceHandle>(a) == static_cast<xclDeviceHandle>(b));
assert(a.get_handle() == b.get_handle());
}
```
both `a` and `b` refer to the same device but the asserts fail.

keryell, 2022-04-20:
@uday610 Is there also `a == b` like in CUDA/OpenCL or CUDA?

uday610, 2022-04-21:
the BDF of the device object should be unique. Maybe you can try that?

```
a.get_info<xrt::info::device::bdf>()
b.get_info<xrt::info::device::bdf>()
```

stsoe, 2022-04-24:
@Ralender They are different device objects referring to the same device.  Under the hood the device is opened twice in your code example and the kernel driver has two device objects.   xrt::device ctor is all but a wrapper around calls to xclOpen().

stsoe, 2022-04-24:
> Is there also `a == b` like in CUDA/OpenCL or SYCL?

There isn't, but there could be.

keryell, 2022-04-24:
At the end this an object representing the same device, so it should work like references on some shared object.
So if you have
```c++
auto a = xrt::device(0);
auto b = xrt::device(0);
auto c = a;
```
they all represent different device values?

If you are looking for some inspiration: :-)
- https://www.khronos.org/registry/SYCL/specs/sycl-2020/html/sycl-2020.html#sec:reference-semantics
- https://github.com/KhronosGroup/SYCL-CTS/blob/SYCL-2020/tests/device/device_constructors.cpp

stsoe, 2022-04-25:
@keryell I am not looking for inspiration, I am looking for a request to implement operator ==(). 

The current behavior is correct;  applications should not dig into the opaque implementation of the device object and draw conclusions from that.  Neither the `get_handle` or the `cast operator` are documented.

Nah, `c = a` is not a new device object; it's an increment of a reference count.  The documentation could be improved to make it clear all xrt::objects are reference counted pimpls.

Ralender, 2022-04-29:
> the BDF of the device objects should be unique. Maybe you can try that?
> 
> ```
> a.get_info<xrt::info::device::bdf>()
> b.get_info<xrt::info::device::bdf>()
> ```

thanks for the tip. it solved the need I had.

uday610, 2022-05-02:
Closed it as original problem is solved

PR #6652 (2022-05-06): Issue #6619 Implement xrt::device equality
Two xrt::device objects referring to same physical device
now compare equal.

#### Problem solved by the commit
Equality fails on two xrt::device objects even if both refer to same underlying physical device

#### Bug / issue (if any) fixed, which PR introduced the bug, how it was discovered
Fixes #6619

#### How problem was solved, alternative solutions (if any) and why they were rejected
Implement operator== to compare the device id with which the device object was constructed.

#### Risks (if any) associated the changes in the commit
Behavior chang [... cut]
closed 2022-05-02 by uday610