Xilinx/XRT #6620: ALVEO U200 verify kernel test failed

pyhundan, 2022-04-20:
Hi,
I've installed an ALVEO U200 and went through the Getting Started guide and all things are well. But when I  compile XRT from the right version source([Official 2020.2 release](https://github.com/Xilinx/XRT/releases/tag/202020.2.8.743)), a problem has appeared. It's an Ubuntu 16.04 system.

The lspci command is as expected:

`#sudo lspci -vd 10ee:`

```
3b:00.0 Processing accelerators: Xilinx Corporation Device 5000
	Subsystem: Xilinx Corporation Device 000e
	Flags: bus master, fast devsel, latency 0
	Memory at 382ff2000000 (64-bit, prefetchable) [size=32M]
	Memory at 382ff4000000 (64-bit, prefetchable) [size=128K]
	Capabilities: [40] Power Management version 3
	Capabilities: [60] MSI-X: Enable+ Count=33 Masked-
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [1c0] #19
	Capabilities: [400] Access Control Services
	Capabilities: [410] #15
	Kernel driver in use: xclmgmt
	Kernel modules: xclmgmt

3b:00.1 Processing accelerators: Xilinx Corporation Device 5001
	Subsystem: Xilinx Corporation Device 000e
	Flags: bus master, fast devsel, latency 0, IRQ 121
	Memory at 382ff0000000 (64-bit, prefetchable) [size=32M]
	Memory at 382ff4020000 (64-bit, prefetchable) [size=64K]
	Memory at 382fe0000000 (64-bit, prefetchable) [size=256M]
	Capabilities: [40] Power Management version 3
	Capabilities: [60] MSI-X: Enable+ Count=33 Masked-
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [400] Access Control Services
	Capabilities: [410] #15
	Kernel driver in use: xocl
	Kernel modules: xocl
```

Besides, the scan is right. 

```
#sudo /opt/xilinx/xrt/bin/xbmgmt flash --scan
```

```
Card [0000:3b:00.0]
    Card type:		u200
    Flash type:		SPI
    Flashable partition running on FPGA:
        xilinx_u200_xdma_201830_2,[ID=0x5d1211e8],[SC=4.2.0]
    Flashable partitions installed in system:
        xilinx_u200_xdma_201830_2,[ID=0x5d1211e8],[SC=4.2.0]
```

The problem output when I tried to validate

`#/opt/xilinx/xrt/bin/xbutil validate`

```
INFO: Found 1 cards

INFO: Validating card[0]: xilinx_u200_xdma_201830_2
INFO: == Starting Kernel version check:
WARNING: Kernel verison 5.1.0 is not officially supported. 5.4.0 is the latest supported version
WARN: == Kernel version check PASSED with warning
INFO: == Starting AUX power connector check:
INFO: == AUX power connector check PASSED
INFO: == Starting Power warning check:
INFO: == Power warning check PASSED
INFO: == Starting PCIE link check:
INFO: == PCIE link check PASSED
INFO: == Starting SC firmware version check:
INFO: == SC firmware version check PASSED
INFO: == Starting verify kernel test:
.....................................................
```

When the steps get `Starting verify kernel test`, it will stop like described above.

Everything was fine until the BUG appeared(the bug is https://support.xilinx.com/s/article/76960?language=en_US), but since the patch, self-compiled XRT has been unable to pass Validate.

Any ideas on what I need to do to fix this?

uday610, 2022-04-21:
Hello,
It is not clear what is the relation between your issue with that y2k patch that is posted by Xilinx technical support team... Maybe you can check with the technical support team by posting this issue in the technical support forum? 

What you mean by self-compiled XRT? Have you tried with XRT package downloading from xilinx website instead of self-compilation?

pyhundan, 2022-04-21:
Hi, @uday610 
      
     I have tried with XRT package downloading from xilinx website. It works successfully. But when I tried to install myself-compilation XRT, the error will appear.

uday610, 2022-04-22:
Hi, if you able to work with XRT package downloading, then you should be able to compile and run with XRT source code which corresponding to the same package. I dont know which version of the source code you are compiling and what XRT package you working with you. But if you say everything works before that Y2K issue, then you should check with Vitis-Support team. Also you can check from other terminal why your xbutil validate is not finishing, if you can do "xbutil query" then it can show CU status, maybe CU is still running/hang, so test is never finishing.

stsoe, 2022-04-23:
@pyhundan You are running into an issue with the ERT firmware because most likely you don't have Vitis installed and XILINX_VITIS defined.  Please see https://xilinx.github.io/XRT/master/html/build.html for details.

pyhundan, 2022-05-09:
Hi, @uday610, the version of the source code I complied is [Official 2020.2 release](https://github.com/Xilinx/XRT/releases/tag/202020.2.8.743)

pyhundan, 2022-05-09:
@stsoe , But I have install Vitis installed before, this problem still occurs.

stsoe, 2022-05-09:
@pyhundan Have you set XILINX_VITIS ?
What does the build output show?

stsoe, 2022-05-18:
Alright sounds good.  In general we don't advise folks to source settings64.sh because it changes the path to tools such as CMake which are sometimes incompatible with the build requirements of XRT.   It is better to simply manual set XILINX_VITIS in a fresh `system` shell without mocking around with the environment.

@pyhundan Please close this issue if your problem is solved.

stsoe, 2022-05-30:
Resolved

closed 2022-05-30 by stsoe