Xilinx/XRT #6637: Unexpected Behavior of clEnqueueNDRangeKernel, clEnqueueTask functions

rogumag, 2022-04-27:
Hi, i was trying to implement a Hardware video algorithm in Alveo U50 card. In this implementation, i have used 3 kernels ( this kernels working in parallel using a OUT_OF_ORDER Queue) working with different buffers. The Main code works as following:
1. Copy the Host PC buffer to the Alveo HBM buffers using clEnqueueMigrateMemObjects functions. The buffer size is only 64KB.
2. The 3 Kernels process the input information (our hardware algorithm method.
3. Finally, copy the output information from the Alveo HBM buffer to Host PC buffer. The buffer size is only 64KB.

This operations are repeated for a certain number of iterations (MAX_LAUNCH). The implemented code is as follows:
```
for (int n = 0; n < MAX_LAUNCH; n++) {
    	err_m = clEnqueueMigrateMemObjects(commands, num_cu,d_memoria_input, 0, 0,  NULL, &wr_event);
    	if (err_m != CL_SUCCESS) printf("Kernel -> clEnqueueMigrateMemObjects H2F return: %d Call Number: %d\n", err,llamadas);
    	for (int i = 0; i < 3; i++) {
    		llamadas++;
    		err = clEnqueueNDRangeKernel(commands, kernel[i], 1, &global_offset, (size_t*)&global, (size_t*)&local, 1, &wr_event, &kernel_event[i]);
    		if (err != CL_SUCCESS) printf("Kernel -> clEnqueueNDRangeKernel return: %d Call Number: %d\n", err,llamadas);
    	}
    	err_m = clEnqueueMigrateMemObjects(commands, num_cu, d_memoria_output, CL_MIGRATE_MEM_OBJECT_HOST, 1,  kernel_event , &rd_event);
    	if (err_m != CL_SUCCESS) printf("Kernel -> clEnqueueMigrateMemObjects F2H return: %d Call Number: %d\n", err,llamadas);
    	clFinish(commands);
}

```
The Issue
If the value of MAX_LAUNCH is less than 21790 , the application works fine. This is the output of application:
```
INFO: Found 1 devices
CL_DEVICE_NAME xilinx_u50_gen3x16_xdma_201920_3
Selected xilinx_u50_gen3x16_xdma_201920_3 as the target device
INFO: Created a command commands OK
INFO: loading xclbin HEVC_3ME_XDMA_PR.xclbin
INFO: kernel loaded from xclbin OK!
The PL was programmed with the kernel
INFO: Build compute program from binary OK!
Host Buffers input/output created
FPGA Buffers input/output created
INFO: Creating a kernel [0] for CU(HEVC_ME:{HEVC_ME_1})
INFO: Set kernel [0] for CU(HEVC_ME:{HEVC_ME_1}) arguments OK!
INFO: Creating a kernel [1] for CU(HEVC_ME:{HEVC_ME_2})
INFO: Set kernel [1] for CU(HEVC_ME:{HEVC_ME_2}) arguments OK!
INFO: Creating a kernel [2] for CU(HEVC_ME:{HEVC_ME_3})
INFO: Set kernel [2] for CU(HEVC_ME:{HEVC_ME_3}) arguments OK!
INFO: Test completed successfully.
Kernel -> clEnqueueNDRangeKernel: 65340 Calls OK of 65340 Total Calls Queued
```
but if the MAX_VALUE is greater than 21780, the application will no longer terminate correctly. This is the output of applicatión with MAX_VALUE=21790:
```
INFO: Found 1 devices
CL_DEVICE_NAME xilinx_u50_gen3x16_xdma_201920_3
Selected xilinx_u50_gen3x16_xdma_201920_3 as the target device
INFO: Created a command commands OK
INFO: loading xclbin HEVC_3ME_XDMA_PR.xclbin
INFO: kernel loaded from xclbin OK!
The PL was programmed with the kernel
INFO: Build compute program from binary OK!
Host Buffers input/output created
FPGA Buffers input/output created
INFO: Creating a kernel [0] for CU(HEVC_ME:{HEVC_ME_1})
INFO: Set kernel [0] for CU(HEVC_ME:{HEVC_ME_1}) arguments OK!
INFO: Creating a kernel [1] for CU(HEVC_ME:{HEVC_ME_2})
INFO: Set kernel [1] for CU(HEVC_ME:{HEVC_ME_2}) arguments OK!
INFO: Creating a kernel [2] for CU(HEVC_ME:{HEVC_ME_3})
INFO: Set kernel [2] for CU(HEVC_ME:{HEVC_ME_3}) arguments OK!
XRT build version: 2.14.0
Build hash: b0f64861167bd808eddb6675c10274512402d3f2
Build date: 2022-04-26 10:20:09
Git branch: master
PID: 45852
UID: 1000
[Wed Apr 27 11:14:11 2022 GMT]
HOST: VLSI-FPGA
EXE: /opt/hardware/HEVC_U50_XDMA_2_HEVC_host/Hardware/HEVC_U50_XDMA_2_HEVC_host
[XRT] ERROR: could not map BO: Invalid argument
Kernel -> clEnqueueNDRangeKernel return: -6 Call Number: 65366
[XRT] ERROR: could not map BO: Invalid argument
Kernel -> clEnqueueNDRangeKernel return: -6 Call Number: 65367
[XRT] ERROR: could not map BO: Invalid argument
Kernel -> clEnqueueNDRangeKernel return: -6 Call Number: 65368
[XRT] ERROR: could not map BO: Invalid argument
Kernel -> clEnqueueNDRangeKernel return: -6 Call Number: 65369
[XRT] ERROR: could not map BO: Invalid argument
Kernel -> clEnqueueNDRangeKernel return: -6 Call Number: 65370
INFO: Test failed
Kernel -> clEnqueueNDRangeKernel: 65365 Calls OK of 65370 Total Calls Queued
```
clEnqueueNDRangeKernel returns CL_OUT_OF_HOST_MEMORY only in the five last calls of the function. The error is the same changing clEnqueueNDRangeKernel for clEnqueueTask.  I'm clueless why this is happening? Is this a bug? 

**SETUP**
Ubuntu 20.04 (5.13.0-40 kernel)
XRT 2.14.0
Vitis 2021.2 
Platform: xilinx_u50_gen3x16_xdma_201920_3 

Regards

stsoe, 2022-04-28:
Hi @rogumag . Your have to manage the OpenCL resources.   I look at the code block you pasted and it is capturing events but not releasing them.  This in turn holds on to a bunch of objects including memory buffers.

rogumag, 2022-04-28:
Hi @stsoe , you are my hero!!!!. The problem has been fixed only releasing the kernel event for each iteration. I close this issue
Thank you so much!!!

closed 2022-04-28 by rogumag