Xilinx/XRT #6643: xrt 2019 error

crizy, 2022-04-28:
I downloaded the xrt version(2019.2 and 2019.2_PU2),And compile and build them respectively By executing ./build.sh. The compiled and built DEB files are as follows,
(1) xrt_201920.2.3.0_18.04-xrt.deb 
(2)xrt_201920.2.5.0_18.04-amd64-xrt.deb

 I install the XRT .deb version  respectively, Then I check their installed version

with sudo apt list | grep XRT command, They display the following version

sudo apt list | grep xrt

xrt-xrt/now 2.5.0 amd64 [installed,local]
xrt-xrt/now 2.3.0 amd64 [installed,local]

but i install the pre-built  xrt versions（xrt_201920.2.5.309_18.04-amd64-xrt.deb/xrt_201920.2.3.1301_18.04-amd64-xrt.deb )  provided by xilinx, Then I check their installed version with sudo apt list | grep XRT command, They display the following version:

sudo apt list | grep xrt

xrt/now 2.5.309 amd64 [installed,local]
xrt/now 2.3.1301 amd64 [installed,local]

I want to know where I need to modify the XRT source code to make the installed XRT display XRT instead of XRT-XRT

stsoe, 2022-04-28:
We no longer support 2019.2.  Regardless, the `xrt-xrt` is wrong package name, it should be just `xrt`.  I have seen this before, but don't recall how it came about.  Can you dig a little deeper?

crizy, 2022-04-28:
> We no longer support 2019.2. Regardless, the `xrt-xrt` is wrong package name, it should be just `xrt`. I have seen this before, but don't recall how it came about. Can you dig a little deeper?

I just git the XRT source code of the above two versions, and then the operation is as follows:

(1)cd build
(2)./build. sh
(3)cd Release
(4)sudo dpkg -i  xrt_201920.2.3.0_18.04-xrt.deb or xrt_201920.2.5.0_18.04-amd64-xrt.deb
(5)sudo apt list | grep xrt

Then the version information of the following XRT installation is displayed
xrt-xrt/now 2.5.0 amd64 [installed,local]
or
xrt-xrt/now 2.3.0 amd64 [installed,local]

crizy, 2022-04-28:
> I just git the XRT source code of the above two versions, and then the operation is as follows:
>
> (1)cd build (2)./build. sh (3)cd Release (4)sudo dpkg -i xrt_201920.2.3.0_18.04-xrt.deb or [...]

then I searched for 'xrt-xrt' in the XRT source code path and got the following results

grep -r 'xrt-xrt'
XRT/build/Release/_CPack_Packages/Linux/DEB/xrt_201920.2.3.1301_18.04/xrt/control:Package: xrt-xrt

crizy, 2022-04-28:
> I just git the XRT source code of the above two versions, and then the operation is as follows:
>
> (1)cd build (2)./build. sh (3)cd Release (4)sudo dpkg -i xrt_201920.2.3.0_18.04-xrt.deb or [...]

crizy, 2022-04-28:
> I just git the XRT source code of the above two versions, and then the operation is as follows:
>
> (1)cd build (2)./build. sh (3)cd Release (4)sudo dpkg -i xrt_201920.2.3.0_18.04-xrt.deb or [...]

then I searched for 'xrt-xrt' in the XRT source code path and got the following results
grep -r 'xrt-xrt'
XRT/build/Release/_CPack_Packages/Linux/DEB/xrt_201920.2.3.0_18.04/xrt/control:Package: xrt-xrt

But I don't know how to change it,  because I'm not engaged in software development

stsoe, 2022-04-28:
Hi @crizy .  I found this PR merged long ago that explains the `xrt-xrt` issue.  It looks like it is a cmake version thing.
https://github.com/Xilinx/XRT/pull/3129

After you successfully build XRT with proper CMake version, then make sure your greps shown only `xrt`.
You will have to make sure no `xrt-xrt` is installed to begin with, .e.g. reset back to no xrt installed before installing properly built XRT.  Hope this will work for you.

crizy, 2022-04-29:
> Hi @crizy . I found this PR merged long ago that explains the `xrt-xrt` issue. It looks like it is a cmake version thing. #3129
>
> After you successfully build XRT with proper CMake version, then make [...]

thanks , 
I check my computer cmake version as follows,
(1)cmake --version
    cmake version 3.3.2
(2)which cmake 
    /opt/Xilinx/Vitis/2019.2/tps/lnx64/cmake-3.3.2/bin/cmake

I found that cmake was installed in xilixnx Vitis software, 
I downloaded, compiled and installed cmake 3.8.2,  cmake 3.8.2 is installed in  /user/local/bin/cmake
How do I upgrade the cmake version from 3.3.3 to 3.8.2?

stsoe, 2022-04-29:
Hi @crizy .  This is good information, thanks.  

You should have to make sure that when you build XRT you do that from a clean shell that is no polluted with Vitis setup.  If your system has CMake >= 3.5 installed when this is the CMake that will be used from a fresh shell.  Make sure 'which cmake` points to your system's cmake.

crizy, 2022-05-05:
> You should have to make sure that when you build XRT you do that from a clean shell that is no polluted with Vitis setup. If your system has CMake >= 3.5 installed when this is the CMake that will be used from a fresh shell. Make sure 'which cmake` points to your system's cmake.

Hi @stsoe,
         Thank you very much for your reply,  now i cmake verison is 3.8.2 in system,But I don't know what I did that caused me to report the following errors when compiling XRT 2019.2
         
         CMakeFiles/xclbintest.dir/XclBinSignature.cxx.o: In function `sk_X509_new_null':
         /usr/include/openssl/x509.h:99: undefined reference to `OPENSSL_sk_new_null'
        CMakeFiles/xclbintest.dir/XclBinSignature.cxx.o: In function `sk_X509_push':
        /usr/include/openssl/x509.h:99: undefined reference to `OPENSSL_sk_push'
       
        src/runtime_src/tools/xclbin/XclBinSignature.cxx:181: undefined reference to `BIO_new_file'
        src/runtime_src/tools/xclbin/XclBinSignature.cxx:181: undefined reference to `BIO_new_file'
        src/runtime_src/tools/xclbin/XclBinSignature.cxx:188: undefined reference to `BIO_new_file'
        src/runtime_src/tools/xclbin/XclBinSignature.cxx:193: undefined reference to `PEM_read_bio_PrivateKey'
        src/runtime_src/tools/xclbin/XclBinSignature.cxx:198: undefined reference to `BIO_free'
        src/runtime_src/tools/xclbin/XclBinSignature.cxx:201: undefined reference to `BIO_new_file'
        src/runtime_src/tools/xclbin/XclBinSignature.cxx:206: undefined reference to `PEM_read_bio_X509'
        src/runtime_src/tools/xclbin/XclBinSignature.cxx:207: undefined reference to `BIO_ctrl'
        src/runtime_src/tools/xclbin/XclBinSignature.cxx:210: undefined reference to `d2i_X509_bio'
        src/runtime_src/tools/xclbin/XclBinSignature.cxx:217: undefined reference to `BIO_free'
        src/runtime_src/tools/xclbin/XclBinSignature.cxx:220: undefined reference to `OPENSSL_init_crypto'
        src/runtime_src/tools/xclbin/XclBinSignature.cxx:221: undefined reference to `EVP_get_digestbyname'
        src/runtime_src/tools/xclbin/XclBinSignature.cxx:231: undefined reference to `CMS_sign'
        src/runtime_src/tools/xclbin/XclBinSignature.cxx:238: undefined reference to `CMS_add1_signer'
        src/runtime_src/tools/xclbin/XclBinSignature.cxx:245: undefined reference to `CMS_final'
        src/runtime_src/tools/xclbin/XclBinSignature.cxx:250: undefined reference to `BIO_free'
        src/runtime_src/tools/xclbin/XclBinSignature.cxx:253: undefined reference to `BIO_s_mem'
        src/runtime_src/tools/xclbin/XclBinSignature.cxx:253: undefined reference to `BIO_new'
        src/runtime_src/tools/xclbin/XclBinSignature.cxx:254: undefined reference to `i2d_CMS_bio_stream'
        src/runtime_src/tools/xclbin/XclBinSignature.cxx:259: undefined reference to `BIO_ctrl'

Judging from the error log, there is something wrong with OpenSSL,

(1)openssl version -a
OpenSSL 1.1.1  11 Sep 2018
built on: Wed Mar  9 12:13:40 2022 UTC
platform: debian-amd64
options:  bn(64,64) rc4(16x,int) des(int) blowfish(ptr) 
compiler: gcc -fPIC -pthread -m64 -Wa,--noexecstack -Wall -Wa,--noexecstack -g -O2 -fdebug-prefix-map=/build/openssl-vxXVMf/openssl-1.1.1=. -fstack-protector-strong -Wformat -Werror=format-security -DOPENSSL_USE_NODELETE -DL_ENDIAN -DOPENSSL_PIC -DOPENSSL_CPUID_OBJ -DOPENSSL_IA32_SSE2 -DOPENSSL_BN_ASM_MONT -DOPENSSL_BN_ASM_MONT5 -DOPENSSL_BN_ASM_GF2m -DSHA1_ASM -DSHA256_ASM -DSHA512_ASM -DKECCAK1600_ASM -DRC4_ASM -DMD5_ASM -DAES_ASM -DVPAES_ASM -DBSAES_ASM -DGHASH_ASM -DECP_NISTZ256_ASM -DX25519_ASM -DPADLOCK_ASM -DPOLY1305_ASM -DNDEBUG -Wdate-time -D_FORTIFY_SOURCE=2
OPENSSLDIR: "/usr/lib/ssl"
ENGINESDIR: "/usr/lib/x86_64-linux-gnu/engines-1.1"
Seeding source: os-specific

(2)which openssl
/usr/bin/openssl

(3) ssh -V
OpenSSH_7.6p1 Ubuntu-4ubuntu0.5, OpenSSL 1.0.2n  7 Dec 2017

How to correct this problem?

stsoe, 2022-05-05:
@crizy  I cannot imagine the dependency script being the reason.  I assume your build was after running `build.sh -clean`? 
Which version of ubuntu are you using?   Now I am simply curious, so I would like to try in a docker container with same version.

crizy, 2022-05-05:
hi @stsoe ,
        yes, I do ./build. sh clean before every build，my ubuntu version is 18.04
        Maybe the XRT source code in my system is confused
        I deleted The source code of XRT 2019.2 is in my system. I  re git  the source code of XRT 2019.2 again and compiled it again. The compilation was successful,thanks

stsoe, 2022-05-05:
Super, thanks @crizy .  Can we close the issue then?

closed 2022-05-05 by crizy