Xilinx/XRT #6672: Repo has mixed line endings

kiwichris, 2022-05-06:
Is this known about and expected?

To see a list run (on FreeBSD): `find . -type f -exec file \{\} \; | grep CRLF`

I am seeing in my clone of the repo the following files with `CRLF`:
```
src/runtime_src/core/edge/sw_em/generic_pcie_hal2/swscheduler.cxx                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                               
src/runtime_src/core/edge/sw_em/generic_pcie_hal2/swscheduler.h
src/runtime_src/core/edge/test/add1_zgem.cpp
src/runtime_src/core/edge/test/add1_zgem_64.cpp
src/runtime_src/core/edge/user/test/xclHALProxy2.cpp
src/runtime_src/core/edge/user/test/xclHALProxy2.h
src/runtime_src/core/edge/user/test/zynq_user_test_add1.cpp
src/runtime_src/core/include/xgq_resp_parser.h
src/runtime_src/core/pcie/driver/aws/kernel/mgmt/10-awsmgmt.rules
src/runtime_src/core/pcie/driver/aws/kernel/mgmt/Makefile
src/runtime_src/core/pcie/driver/aws/kernel/mgmt/mgmt-cw.h
src/runtime_src/core/pcie/driver/linux/xocl/subdev/nifd.c
src/runtime_src/core/pcie/driver/linux/xocl/subdev/ulite.c
src/runtime_src/core/pcie/driver/linux/xocl/userpf/99-xocl.rules
src/runtime_src/core/pcie/emulation/cpu_em/generic_pcie_hal2/swscheduler.cxx
src/runtime_src/core/pcie/emulation/cpu_em/generic_pcie_hal2/swscheduler.h
src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/mbscheduler.cxx
src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/mbscheduler.h
src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/mbscheduler_hwemu.cxx
src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/mbscheduler_hwemu.h
src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/xgq_hwemu.cxx
src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/xgq_hwemu.h
src/runtime_src/core/tools/common/OptionOptions.cpp
src/runtime_src/core/tools/common/OptionOptions.h
src/runtime_src/core/tools/common/Process.cpp
src/runtime_src/core/tools/common/Process.h
src/runtime_src/core/tools/common/ProgressBar.cpp
src/runtime_src/core/tools/common/ProgressBar.h
src/runtime_src/core/tools/common/Report.cpp
src/runtime_src/core/tools/common/Report.h
src/runtime_src/core/tools/common/ReportAie.cpp
src/runtime_src/core/tools/common/ReportAie.h
src/runtime_src/core/tools/common/ReportAieShim.cpp
src/runtime_src/core/tools/common/ReportAieShim.h
src/runtime_src/core/tools/common/ReportAsyncError.cpp
src/runtime_src/core/tools/common/ReportAsyncError.h
src/runtime_src/core/tools/common/ReportBOStats.cpp
src/runtime_src/core/tools/common/ReportBOStats.h
src/runtime_src/core/tools/common/ReportCmcStatus.cpp
src/runtime_src/core/tools/common/ReportCmcStatus.h
src/runtime_src/core/tools/common/SubCmd.cpp
src/runtime_src/core/tools/common/ReportDynamicRegion.h
src/runtime_src/core/tools/common/ReportElectrical.cpp
src/runtime_src/core/tools/common/ReportElectrical.h
src/runtime_src/core/tools/common/ReportFirewall.cpp
src/runtime_src/core/tools/common/ReportFirewall.h
src/runtime_src/core/tools/common/ReportHost.cpp
src/runtime_src/core/tools/common/ReportHost.h
src/runtime_src/core/tools/common/ReportMailbox.cpp
src/runtime_src/core/tools/common/ReportMailbox.h
src/runtime_src/core/tools/common/ReportMechanical.cpp
src/runtime_src/core/tools/common/ReportMechanical.h
src/runtime_src/core/tools/common/ReportMemory.cpp
src/runtime_src/core/tools/common/ReportMemory.h
src/runtime_src/core/tools/common/ReportPcieInfo.cpp
src/runtime_src/core/tools/common/ReportPcieInfo.h
src/runtime_src/core/tools/common/ReportPlatforms.cpp
src/runtime_src/core/tools/common/ReportPlatforms.h
src/runtime_src/core/tools/common/ReportQspiStatus.cpp
src/runtime_src/core/tools/common/ReportQspiStatus.h
src/runtime_src/core/tools/common/ReportThermal.cpp
src/runtime_src/core/tools/common/ReportThermal.h
src/runtime_src/core/tools/common/SubCmd.h
src/runtime_src/core/tools/common/XBHelpMenus.cpp
src/runtime_src/core/tools/common/XBHelpMenus.h
src/runtime_src/core/tools/common/XBHelpMenusCore.cpp
src/runtime_src/core/tools/common/XBHelpMenusCore.h
src/runtime_src/core/tools/common/XBMain.cpp
src/runtime_src/core/tools/common/XBMain.h
src/runtime_src/core/tools/common/XBUtilities.h
src/runtime_src/core/tools/common/XBUtilitiesCore.h
src/runtime_src/core/tools/xbflash2/XBFMain.cpp
src/runtime_src/core/tools/xbflash2/XBFMain.h
src/runtime_src/core/tools/xbmgmt2/ReportPlatform.cpp
src/runtime_src/core/tools/xbmgmt2/ReportPlatform.h
src/runtime_src/core/tools/xbmgmt2/SubCmdDump.cpp
src/runtime_src/core/tools/xbmgmt2/SubCmdDump.h
src/runtime_src/core/tools/xbmgmt2/flash/xgq_vmr.cpp
src/runtime_src/core/tools/xbmgmt2/flash/xgq_vmr.h
src/runtime_src/core/tools/xbmgmt2/flash/xospiversal.cpp
src/runtime_src/core/tools/xbmgmt2/flash/xospiversal.h
src/runtime_src/core/tools/xbmgmt2/xbmgmt.bat
src/runtime_src/core/tools/xbmgmt2/xbmgmt.cpp
src/runtime_src/core/tools/xbutil2/xbutil.bat
src/runtime_src/tools/scripts/header.lic
src/runtime_src/tools/scripts/loader.bat
src/runtime_src/tools/scripts/sprite.py
src/runtime_src/tools/scripts/user.lic
src/runtime_src/tools/xclbinutil/unittests/SmartNic/eBPF.json
src/runtime_src/tools/xclbinutil/unittests/SmartNic/extension.json
src/runtime_src/tools/xclbinutil/unittests/SmartNic/softhub.json
src/runtime_src/tools/xclbinutil/unittests/SmartNic/vpp_linker.json
src/runtime_src/tools/xclbinutil/xclbinutil.bat
src/runtime_src/xdp/profile/plugin/noc/noc_cb.cpp
src/runtime_src/xdp/profile/writer/aie_profile/aie_writer.cpp
src/runtime_src/xdp/profile/writer/aie_profile/aie_writer.h
src/runtime_src/xdp/profile/writer/noc/noc_writer.h
src/xma/include/app/xmadecoder.h
src/xma/include/plg/xmadecoder.h
src/xma/src/xmaapi/xmadecoder.cpp
tests/python/23_bandwidth/versal_23_bandwidth.py
```

keryell, 2022-05-06:
Of course not.
Now you have the recipe to detect the problem, you have almost the recipe to make a PR. :-)

kiwichris, 2022-05-06:
Excellent and yes can do that.

Thanks.

stsoe, 2022-05-11:
@kiwichris Thanks for the fix in #6681 .
See also #6687 which should prevent this mess in the future.

closed 2022-05-11 by stsoe