Xilinx/XRT #6714: xrt install  problem

crizy, 2022-05-19:
i Compiled and installed xrt_2019.2

   (1)sudo /opt/xilinx/xrt/bin/xbmgmt flash --scan    is ok ,  it can recognize shell on FPGA and ubuntu

   (2)but sudo /opt/xilinx/xrt/bin/xbutil query  is error , The error message is as follows
               INFO: Found total 1 card(s), 0 are usable
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
               System Configuration
               OS name:	Linux
               Release:	4.15.0-041500-generic
               Version:	#201802011154 SMP Thu Feb 1 11:55:45 UTC 2018
               Machine:	x86_64
               Glibc:		2.27
               Distribution:	Ubuntu 18.04.6 LTS
               Now:		Thu May 19 09:19:34 2022
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
               XRT Information
               Version:	2.3.1301
               Git Hash:	192e706aea53163a04c574f9b3fe9ed76b6ca471
               Git Branch:	HEAD
               Build Date:	2022-05-18 13:33:02
               XOCL:		2.3.1301,192e706aea53163a04c574f9b3fe9ed76b6ca471
              XCLMGMT:	2.3.1301,192e706aea53163a04c574f9b3fe9ed76b6ca471
              ERROR: Card index 0 is out of range

   (3) make all TARGET=sw_emu PLATFORM=xpfm   is ok
        make check/test TARGET=sw_emu PLATFORM=xpfm   is ok  print message as follows,
               Device[0]: program successful!
               TEST PASSED

   (4)  make all TARGET=hw PLATFORM=xpfm   is ok
         make check/test TARGET=hw PLATFORM=xpfm   is error  print message as follows,
               [XRT] WARNING: No devices found
               [XRT] ERROR: No devices found
               [XRT] ERROR: No devices found
               [XRT] ERROR: No devices found
                ..//common/includes/xcl2/xcl2.cpp:46 Error calling err = cl::Platform::get(&platforms), error code is: -1001
               Makefile:141: recipe for target 'test' failed
               make: *** [test] Error 1

the error message calling err = cl::Platform::get(&platforms), error code is: -1001 It looks like a pyopencl installation problem

I follow the following method
 https://support.xilinx.com/s/article/73055?language=en_US ，
However, the above error still exist          

my pip list as follows,
 pip list

DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
WARNING: The directory '/home/ubuntu/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Package       Version
------------- --------
appdirs       1.4.4
asn1crypto    0.24.0
cryptography  2.1.4
decorator     4.4.2
defusedxml    0.5.0
enum34        1.1.6
idna          2.6
iniparse      0.4
ipaddress     1.0.17
keyring       10.6.0
keyrings.alt  3.0
Mako          1.1.6
MarkupSafe    1.1.1
meld3         1.0.2
numpy         1.16.6
pip           20.1.1
psutil        5.4.2
pybind11      2.8.1
pycairo       1.16.2
pycrypto      2.6.1
pygobject     3.26.1
pyliblzma     0.5.3
pyopencl      2019.1.2
pytools       2020.1
pyxdg         0.25
SecretStorage 2.3.1
setuptools    0.9.8
six           1.11.0
SOAPpy        0.12.22
supervisor    3.3.1
wheel         0.30.0
wstools       0.4.3
WARNING: You are using pip version 20.1.1; however, version 20.3.4 is available.
You should consider upgrading via the '/usr/bin/python -m pip install --upgrade pip' command.

uday610, 2022-05-25:
Hello, 2019.2 is 2+ year-old release for debugging. Why don't you move to the latest releases? 
But maybe some basic questions:
1) What card is this? What platform version is this? 
2) Have you already installed platform packages? And flashed the card? In other words, is your card ready? 
3) Assuming you have installed platform package in your system, and flashing is done, then you can reboot the machine and see if you can see xbutil query working. 
4) You can also provide the full dmesg

crizy, 2022-05-28:
> Hello, 2019.2 is 2+ year-old release for debugging. Why don't you move to the latest releases? But maybe some basic questions:
>
> 1. What card is this? What platform version is this?
> 2. Have you [...]

@uday610
   i build   https://github.com/Xilinx/XRT/releases/tag/2019.2_RC1
   build is ok,
   but  path build/Release/opt/xilinx/xrt/share/   is no fw/sched*.bin,

stsoe, 2022-05-30:
@crizy Please check #6620.
Make sure XILINX_VITIS is set in the shell used to build XRT.

crizy, 2022-05-31:
> @crizy Please check #6620. Make sure XILINX_VITIS is set in the shell used to build XRT.

hi @stsoe ，

     before i build xrt. i install vitis/vivado and set  vitis/vivado/xrt environment variables in .bashrc  as follows

             source /opt/Xilinx/Vivado/2019.2/settings64.sh
             source /opt/Xilinx/Vitis/2019.2/settings64.sh
             source /opt/xilinx/xrt/setup.sh
             export LIBRARY_PATH=/usr/lib/x86_64-linux-gnu
             export PLATFORM_REPO_PATHS=/opt/xilinx/platforms

stsoe, 2022-05-31:
I would recommend just setting XILINX_VITIS in a fresh shell, and not sourcing any environment changing scripts.  In particular Vitis' settings64 is not compatible with a system shell, I believe it changes PATH and ends up using tools / libraries that should not be used.

May not work either way, and if not maybe let me know which OS you are on.

crizy, 2022-05-31:
> I would recommend just setting XILINX_VITIS in a fresh shell, and not sourcing any environment changing scripts. In particular Vitis' settings64 is not compatible with a system shell, I believe it [...]

hi @stsoe ，
thanks for you reply
my os is ubuntu18.04, i set vitis/vivado/xrt environment variables in .bashrc 

It is better to simply manual set XILINX_VITIS in a fresh system shell without mocking around with the environment.
how to manual set XILINX_VITIS in a fresh system shell?

others,  when i build xrt_2019_pu2 with same operate ,it 's ok , path build/Release/opt/xilinx/xrt/share/ has  fw/sched*.bin,
i find build.sh is different between xrt_2019_pu2(has ert parameter) and  2019.2_RC1

stsoe, 2022-05-31:
> how to manual set XILINX_VITIS in a fresh system shell?

export XILINX_VITIS=<path>

> others, when i build xrt_2019_pu2 with same operate ,it 's ok , path build/Release/opt/xilinx/xrt/share/ has fw/sched*.bin, i find build.sh is different between xrt_2019_pu2(has ert parameter) and 2019.2_RC1

Sorry I don't follow?  Can you show the difference?  We don't really support 2019.2 at this point.  Very old.

crizy, 2022-05-31:
> export XILINX_VITIS=

Compared with 2019.2_RC1 , xrt_2019_pu2 the following parameters in build.sh
[image]
[image]

but when i replace2019.2_RC1'build.sh with xrt_2019_pu2'build.sh, after build, 2019.2_RC1 path build/Release/opt/xilinx/xrt/share/ is still  no  fw/sched*.bin

stsoe, 2022-05-31:
@crizy The build.sh changes are not relevant for what you are seeing.
I clone 2019.2_RC1 and I can see that the firmware builds are not officially supported in that release.
In other words, you cannot create the ERT firmware in that release.  
I suggest you download the archived Debian 18.04 package from Xilinx and use the firmware included in that build
* https://www.xilinx.com/products/boards-and-kits/alveo/package-files-archive/u200-2019-2.html

crizy, 2022-06-06:
> @crizy The build.sh changes are not relevant for what you are seeing. I cloned 2019.2_RC1 and I can see that the firmware builds are not officially supported in that release. In other words, you cannot create the ERT firmware in that release. I suggest you download the archived Debian 18.04 package from Xilinx and use the firmware included in that build
> 
> * https://www.xilinx.com/products/boards-and-kits/alveo/package-files-archive/u200-2019-2.html

hi @stsoe ，

2019.2_RC1'hash is same as xrt_2.3.1301(https://www.xilinx.com/products/boards-and-kits/alveo/package-files-archive/u200-2019-2.html), The following quotation is your answer to my question(https://github.com/Xilinx/XRT/issues/6514#event-6390256873)

Hi @crizy . You can check for the git hash in /opt/xilinx/xrt/include/version.h or /opt/xilinx/xrt/version.json.
The source code is here:
2.3.1301,192e706aea53163a04c574f9b3fe9ed76b6ca471

stsoe, 2022-06-06:
Yes, that is good.  What's the question?  Can we close this issue?

closed 2022-06-07 by crizy