Xilinx/XRT #6779: Why xbflash installed to /usr/local/bin ?

wilderfield, 2022-06-17:
Most XRT utilities are installed to XRT_INSTALL_DIR/bin

Why does xbflash break from the norm, and use hardcoded path `/usr/local/bin`?

https://github.com/Xilinx/XRT/blame/master/src/runtime_src/core/pcie/tools/xbflash.qspi/CMakeLists.txt#L18

xuhz, 2022-06-17:
xbflash.qspi was introduced to be used without dependency on xrt. When xrt is built, there is a separate pkg built for xbflash.qspi and this pkg can be installed alone without xrt being installed first
this is not issue

closed 2022-06-17 by xuhz

stsoe, 2022-06-18:
I agree that it should be installed like other components under /opt/xilinx/xrt. 
Like I mentioned earlier, all of XRT should be installed relative to /usr/ so that no set up scripts are needed at all.

bryanloz-xilinx, 2022-06-18:
> I agree that it should be installed like other components under /opt/xilinx/xrt.
> Like I mentioned earlier, all of XRT should be installed relative to /usr/ so that no set up scripts are needed at all.

@xuhz want to reopen this?

@stsoe I can work on a new PR for that, but I'm a little concerned.

/opt/xilinx is hardcoded inside of 174 locations in the XRT repo inside of 56 different files.

```
$ egrep -r ".opt.xilinx" . -l
./src/CMakeLists.txt
./src/CMake/config/postinst.in
./src/CMake/config/postinst-container.in
./src/CMake/config/prerm-container.in
./src/CMake/config/postinst-aws.in
./src/CMake/config/prerm-aws.in
./src/CMake/config/prerm-azure.in
./src/CMake/config/prerm.in
./src/CMake/config/postinst-azure.in
./src/runtime_src/tools/xclbinutil/unittests/PSKernel/pskernel.cpp
./src/runtime_src/tools/scripts/setup.sh
./src/runtime_src/tools/scripts/setup.csh
./src/runtime_src/tools/scripts/README.md
./src/runtime_src/tools/scripts/install.sh
./src/runtime_src/tools/scripts/pkg_hw_platform.sh
./src/runtime_src/tools/scripts/pkgdsa.sh
./src/runtime_src/core/tools/common/Process.cpp
./src/runtime_src/core/tools/xbmgmt2/flash/flasher.cpp
./src/runtime_src/core/tools/xbmgmt2/flash/firmware_image.h
./src/runtime_src/core/tools/xbutil2/SubCmdValidate.cpp
./src/runtime_src/core/pcie/tools/cloud-daemon/msd.cpp
./src/runtime_src/core/pcie/tools/cloud-daemon/mpd.cpp
./src/runtime_src/core/pcie/tools/cloud-daemon/container/container.cpp
./src/runtime_src/core/pcie/tools/cloud-daemon/msd_plugin_example.c
./src/runtime_src/core/edge/ps_kernels/sample/pskernel.cpp
./src/runtime_src/xdp/CMakeLists.txt
./src/runtime_src/ert/CMakeLists.txt
./src/runtime_src/doc/toc/yocto.rst
./src/runtime_src/doc/toc/cloud_vendor_support.rst
./src/runtime_src/doc/toc/build.rst
./src/xma/src/xmaapi/xma_utils.cpp
./CMakeLists.txt
./build/board.sh
./build/run.sh
./build/build.sh
./build/debian/xrt-xocl-dkms.postinst
./build/debian/rules
./build/debian/xrt-xocl-dkms.prerm
./tests/message-mgmt/Makefile
./tests/unit_test/subdevice/readme.txt
./tests/unit_test/cuselect/readme.txt
./tests/unit_test/dataflow/main.cpp
./tests/xrt/perf_IOPS/README.md
./tests/xma/check_xmafilter/Makefile
./tests/xma/check_xmaenc/Makefile
./tests/xma/check_xmacfg/Makefile
./tests/xma/check_xmascaler/Makefile
./tests/xma/check_xmakernel/Makefile
./tests/xma/check_xmaapi/Makefile
./tests/xma/check_xmadec/Makefile
./tests/xma/plugins/Makefile
./tests/xma/check_xmalogger/Makefile
./tests/xma/check_xmabuffer/Makefile
./tests/xma/xma_tests.sh
./.github/scripts/pipeline/install_platform.sh
./.github/workflows/pipeline.yml
```

stsoe, 2022-06-19:
Yeah,  I don't think we can change this now.  At most we can release to /opt/xilinx/ also.
While I agree with /opt/xilinx/, I wonder why /usr/local/bin is a problem to begin with?
I don't think we should re-open unless it is an issue, is it, why?

bryanloz-xilinx, 2022-06-19:
> Yeah, I don't think we can change this now. At most we can release to /opt/xilinx/ also.
> While I agree with /opt/xilinx/, I wonder why /usr/local/bin is a problem to begin with?
> I don't think we should re-open unless it is an issue, is it, why?
@xuhz @stsoe 

Since I have had the pleasure of building XRT on Linux, Windows, inside Docker, inside Anaconda, in my sleep... I have also had the pleasure of building it's dependencies which have become ubiquitous in the industry. Dependencies such as boost, protobuf, json-c, etc...

**ALL** of these dependencies allow their end users to install the package to the directory of their choosing. They do not hardcode installation paths. The purpose of `CMAKE_INSTALL_PREFIX` is to avoid hardcoding installation paths, and allow packages to be relocatable.

Allowing end users to install wherever they want is very valuable, especially for people who are trying to work with XRT who do not have admin/sudo privileges... maybe they want to install XRT to their home directory?

In my view, it is a capitol sin to hardcode any installation paths for binary executables and/or libraries.

timsnyder-siv, 2022-06-19:
I have been working on a conda-forge recipe for XRT and have some patches that I could clean up and share.  Seems like @bryanloz-xilinx is all over this though.

My current recipe hasn't been submitted to conda-forge yet.  I want to do more local testing and I only have the features I currently need working (haven't included the AWS plugin yet).

bryanloz-xilinx, 2022-06-19:
@timsnyder-siv this has been my weekend project:
https://github.com/conda-forge/staged-recipes/pull/19337

I'm working with the community to cleanup the recipe as we speak.

If you would like to join me as a maintainer of the recipe I would love that.

timsnyder-siv, 2022-06-19:
Sure.  I'd be happy to.  My recipe is at https://github.com/timsnyder/staged-recipes/tree/xilinx-xrt but I got sidetracked this past week with Vitis->Vivado constraint shinnanigans and didn't get through testing it.  I'll take a look at your recipe PR and would be happy to help out as a maintainer as my personal account @timsnyder.

bryanloz-xilinx, 2022-06-19:
> Sure. I'd be happy to. My recipe is at https://github.com/timsnyder/staged-recipes/tree/xilinx-xrt but I got sidetracked this past week with Vitis->Vivado constraint shinnanigans and didn't get [...]

@timsnyder-siv wow! very very interesting! The world is such a small place... cool to see that we both saw the need... and it looks like our approaches are very similar. You already fixed some things with the run exports that I was yet to find a solution for.

stsoe, 2022-06-19:
> 

Once again I completely agree with you.  
CMAKE_INSTALL_PREFIX should be honored, but I doubt that is specific to xbflash?

@bryanloz-xilinx, please open a new issue and complain that CMAKE_INSTALL_PREFIX is not honored as it should be with xbflash as an example.   I believe I recently fixed XRT to use CMAKE_INSTALL_PREFIX in general because I needed to be able to control install area  when building XRT as a sub-module in another project.  I think for that project I think I only needed and built a small subset of XRT.
 I will take ownership of the issue if you file it, but will not promise a quick fix.

bryanloz-xilinx, 2022-06-19:
> Once again I completely agree with you. CMAKE_INSTALL_PREFIX should be honored, but I doubt that is specific to xbflash?
>
> @bryanloz-xilinx, please open a new issue and complain that [...]

I love complaining!

bryanloz-xilinx, 2022-07-02:
@xuhz the problem is that, if someone installs XRT into the system, with: `make install`, xbflash will be installed to `/usr/local/bin`.

```
$ pwd
/home/bryanloz/XRT/build/Release
$ find . -name "xbflash" 
./runtime_src/core/pcie/tools/xbflash.qspi/xbflash
./usr/local/bin/xbflash
```

If someone is trying to install XRT, into an arbitrary directory, your hardcoded installation path is a problem.

You are right that if debians are being used, there is no problem, but debians should not be the only supported use mode.

```
$ dpkg -c xrt_202210.2.13.466_18.04-amd64-xrt.deb | grep xbflash
-rwxr-xr-x root/root    143416 2022-04-14 17:45 ./opt/xilinx/xrt/bin/xbflash.qspi
```