Xilinx/XRT #6782: Anaconda Support from Open Source Community

bryanloz-xilinx, 2022-06-19:
@bryanloz-xilinx and @timsnyder are working with the open source community to create a conda recipe for XRT.

This issue is here to track discussion on that topic.

Here is the staged recipe:

https://github.com/conda-forge/staged-recipes/pull/19337

bryanloz-xilinx, 2022-06-19:
@timsnyder My focus for the first rendition of a conda recipe was user space binaries and libraries.

I figured that a user can separately install XRT from deb or rpm to install drivers and shells.

For this reason, my conda patch removes any dkms related compilation, and cloud-daemon. Anything that interacts with the kernel.

It looks like in your recipe, you are trying to support things that interact with the kernel.

How do you feel about the approach of only focusing on user space for the first recipe?

timsnyder, 2022-06-19:
Doing only user space for the first cut seems reasonable. Probably want the
system stuff to be a seperate package even if it is eventually supported. I
was nervous about compatability between user space code and the kernel
modules and daemons.

timsnyder, 2022-06-20:
@bryanloz-xilinx , looking at your recipe, I think the biggest issue for me is needing to have a version that I'm confident will work with on Azure NP10 VMs with U250 boards.  That's why I've been using the version in my recipe.  We also need the MMIO performance fix that is cherry-picked in one of my patches.  I can circle back with the MS folks or perhaps you know more about this situation being on the inside of things?  This doesn't really seem germane to the quality/readiness of your recipe, hence my comment here.

bryanloz-xilinx, 2022-06-20:
> @bryanloz-xilinx , looking at your recipe, I think the biggest issue for me is needing to have a version that I'm confident will work with on Azure NP10 VMs with U250 boards. That's why I've been [...]

My goal has been to create a recipe for the most recent release. I noticed your recipe was release 2.11 and mine is 2.13. I hope your referenced MIMO fix has made it in. Is this something you can verify?

timsnyder, 2022-06-20:
The MMIO fix I refer to is https://github.com/Xilinx/XRT/commit/63269b8d4aa04099459c68b283f8512748fb39d6 and it seems to be in [202210.2.13.466](https://github.com/Xilinx/XRT/releases/tag/202210.2.13.466)

However, I am concerned that the following pages from Microsoft and Xilinx only talk about using 2.11 on Azure:
* https://docs.microsoft.com/en-us/azure/virtual-machines/np-series#frequently-asked-questions
* https://www.xilinx.com/products/boards-and-kits/alveo/cloud-solutions/microsoft-azure.html

Like I said, I'll ping my contacts at MS and if you or anyone else at Xilinx knows about moving forward to 2.13, I'd be happy to try to push the boundaries.  

In the worst case, we can create a 2.11 branch of the recipe after you get it merged for the latest version if I need to keep using the older one for awhile.

bryanloz-xilinx, 2022-06-20:
> The MMIO fix I refer to is [63269b8](https://github.com/Xilinx/XRT/commit/63269b8d4aa04099459c68b283f8512748fb39d6) and it seems to be in [202210.2.13.466](https://github.com/Xilinx/XRT/releases/tag/202210.2.13.466)
> 
> However, I am concerned that the following pages from Microsoft and Xilinx only talk about using 2.11 on Azure:
> 
> * https://docs.microsoft.com/en-us/azure/virtual-machines/np-series#frequently-asked-questions
> * https://www.xilinx.com/products/boards-and-kits/alveo/cloud-solutions/microsoft-azure.html
> 
> Like I said, I'll ping my contacts at MS and if you or anyone else at Xilinx knows about moving forward to 2.13, I'd be happy to try to push the boundaries.
> 
> In the worst case, we can create a 2.11 branch of the recipe after you get it merged for the latest version if I need to keep using the older one for awhile.

Sounds good.

I’m minorly concerned about the fact that I’m omitting the cloud_daemon when you are on Azure.

We should probably verify that it can be installed separately, and all communication works. I think it should.

bryanloz-xilinx, 2022-07-09:
https://anaconda.org/conda-forge/xilinx-runtime

https://github.com/conda-forge/xilinx-runtime-feedstock

Done!

closed 2022-07-09 by bryanloz-xilinx