Xilinx/XRT #6783: CMAKE_INSTALL_PREFIX is not honored

bryanloz-xilinx, 2022-06-19:
The XRT package is not relocatable.

https://github.com/Xilinx/XRT/blob/master/src/CMakeLists.txt#L69

bryanloz-xilinx, 2022-06-21:
@gbuildx @NaAl-Xlnx 
Can you please take a look at the build pipelines and make sure that `/opt/xilinx` or `/opt/xilinx/xrt` is not hardcoded anywhere in the build pipeline or in any automated tests?

In the near future we would like to install XRT to `/usr/` by default.

qberthet, 2023-10-23:
@stsoe what is the status of this issue? In the discussion under #6779 you mention that you want to address this issue but "will not promise a quick fix".

I would like to weight in that XRT not being relocatable is a problem for the large project I work on. Knowing when we could expect a fix for this could be of great help. Thanks!

qberthet, 2024-04-09:
Now that  #7835 is merged, shouldn't this issue be closed ? Or is there still something else to it?
Thanks!

stsoe, 2024-04-09:
Thanks @qberthet.  The PR was missing meta comment for closing the issue.

closed 2024-04-09 by stsoe