Xilinx/XRT #6786: xrt error question

crizy, 2022-06-20:
i build and install xrt_2019.2 on ubuntu 18.04, 
(1) lspci -vvv -d 10ee:    
     xclmgmt and xocl kernel driver can be recognized

(2) sudo /opt/xilinx/xrt/bin/xbmgmt flash --scan  
     Flashable partition running on FPGA and Flashable partition running on FPGA can be recognized

(3) run Vitis_Accel_Examples_2019/hello_word  example
      sw_emu/hw_emu  is ok (make all and make check are ok)
      make all hw is also ok,but make check hw is error, error information as follows,
              XRT build version: 2.3.1301
              Build hash: 192e706aea53163a04c574f9b3fe9ed76b6ca471
              Build date: 2022-06-15 11:16:00
              Git branch: HEAD
              PID: 159059
              UID: 1000
              [Mon Jun 20 09:59:27 2022]
              HOST: ubuntu-wf
              EXE: /home/ubuntu/git_file/vitis_examples_2019_2/Vitis_Accel_Examples/hello_world/host
              [XRT] WARNING: No devices found
              [XRT] ERROR: No devices found
              [XRT] ERROR: No devices found
              [XRT] ERROR: No devices found
              ..//common/includes/xcl2/xcl2.cpp:46 Error calling err = cl::Platform::get(&platforms), error code is: -1001
              Makefile:143: recipe for target 'test' failed
              make: *** [test] Error 1
(4) sudo /opt/xilinx/xrt/bin/xbutil status
         INFO: Found total 1 card(s), 0 are usable
         ERROR: Card index 0 is out of range

(5)sudo /opt/xilinx/xrt/bin/xbutil query
INFO: Found total 1 card(s), 0 are usable
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
System Configuration
OS name:	Linux
Release:	4.15.0-041500-generic
Version:	#201802011154 SMP Thu Feb 1 11:55:45 UTC 2018
Machine:	x86_64
Glibc:		2.27
Distribution:	Ubuntu 18.04.6 LTS
Now:		Mon Jun 20 15:24:16 2022
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
XRT Information
Version:	2.3.1301
Git Hash:	192e706aea53163a04c574f9b3fe9ed76b6ca471
Git Branch:	HEAD
Build Date:	2022-06-15 11:16:00
XOCL:		2.3.1301,192e706aea53163a04c574f9b3fe9ed76b6ca471
XCLMGMT:	2.3.1301,192e706aea53163a04c574f9b3fe9ed76b6ca471
ERROR: Card index 0 is out of range

(6)clinfo
XRT build version: 2.3.1301
Build hash: 192e706aea53163a04c574f9b3fe9ed76b6ca471
Build date: 2022-06-15 11:16:00
Git branch: HEAD
PID: 85670
UID: 1000
[Mon Jun 20 15:25:00 2022]
HOST: ubuntu-wf
EXE: /usr/bin/clinfo
[XRT] ERROR: No devices found
[XRT] ERROR: No devices found
[XRT] ERROR: No devices found
Number of platforms                               0

(7）vi etc/OpenCL/vendors/xilinx.icd 
/opt/xilinx/xrt/lib/libxilinxopencl.so
(8）ls /opt/xilinx/xrt/lib/libxilinxopencl.so
/opt/xilinx/xrt/lib/libxilinxopencl.so

(9)pip list 
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
WARNING: The directory '/home/ubuntu/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Package       Version
------------- --------
appdirs       1.4.4
asn1crypto    0.24.0
cryptography  2.1.4
decorator     4.4.2
defusedxml    0.5.0
enum34        1.1.6
idna          2.6
iniparse      0.4
ipaddress     1.0.17
keyring       10.6.0
keyrings.alt  3.0
Mako          1.1.6
MarkupSafe    1.1.1
meld3         1.0.2
numpy         1.16.6
pip           20.3.4
psutil        5.4.2
pybind11      2.8.1
pycairo       1.16.2
pycrypto      2.6.1
pygobject     3.26.1
pyliblzma     0.5.3
pyopencl      2019.1.2
pytools       2020.1
pyxdg         0.25
SecretStorage 2.3.1
setuptools    0.9.8
six           1.11.0
SOAPpy        0.12.22
supervisor    3.3.1
wheel         0.30.0
wstools       0.4.3

I found some related question posts on the support forum on the xilinx official website, as follows, But it doesn't seem to have been solved
(1)https://support.xilinx.com/s/article/73055?language=en_US
(2)https://support.xilinx.com/s/article/73440?language=en_US      
(3)https://support.xilinx.com/s/question/0D52E00006hpM40SAE/alveo-u50-xbutil-validate-failed?language=en_US
(4)https://support.xilinx.com/s/question/0D52E00006hpKLISA2/alveo-u200-verify-kernel-test-failed?language=en_US

The following link seems to have a solution, but the solution page is missing....
https://support.xilinx.com/s/question/0D52E00006hpKXASA2/first-alveo-u280-kernel-run-xrt-error-no-devices-found?language=en_US&t=1655704503345

How should I deal with this  cl::Platform::get(&platforms), error code is: -1001 and ERROR: No devices found ? thanks

uday610, 2022-06-21:
Hello, you are building XRT that is 3 years old (as you said xrt_2019.2). It is hard to get support for those old XRT release because things moves fast in last 3 years. It is possible that old XRT release cannot work the card you are using. Which card you are using and which platform?

You dont need to run your hello world example. When xbutil query is showing no device found, then you cannot do anything further. There is no need to run application. 

With that old release and without knowing which card/platform you are using I think it is difficult to get support of this issue. You can try in Xilinx Forum though. You can also troubleshoot from the Alveo Debug Guide: https://xilinx.github.io/Alveo-Cards/master/debugging/build/html/docs/card-not-recognized.html

crizy, 2022-06-22:
> Hello, you are building XRT that is 3 years old (as you said xrt_2019.2). It is hard to get support for those old XRT release because things moves fast in last 3 years. It is possible that old XRT release cannot work the card you are using. Which card you are using and which platform?
> 
> You dont need to run your hello world example. When xbutil query or xbutil status is showing no device found, then you cannot do anything further. There is no need to run application (hello world or otherwise).
> 
> With that old release and without knowing which card/platform you are using I think it is difficult to get support of this issue. You can try in Xilinx Forum though. You can also troubleshoot from the Alveo Debug Guide: https://xilinx.github.io/Alveo-Cards/master/debugging/build/html/docs/card-not-recognized.html

Because the Xilinx tool I installed on ubuntu was vitis/vivado 2019.2, and many of my projects are built with viviado 2019,So I don't want to upgrade the tool