Xilinx/XRT #6827: Buffers flagged as P2P can't be read from / written by host via DMA

moazin, 2022-06-30:
Buffers that are P2P are painfully slow to read and write currently. This is because the transfer is not happening via the DMA.

Please see the chart below.
[image]

@maxzhen @uday610

maxzhen, 2022-08-22:
We have taken a look at this issue and the current thinking is to use xclReadBO and xclWriteBO to move data to/from P2P buffer via DMA. But, we have not got cycles to change the driver to support it. I believe that we need to change the code in xocl_pread_bo_ioctl() and xocl_pwrite_bo_ioctl() to make sure, for P2P BO, we go through xocl_migrate_unmgd(), instead of copy_to_user/copy_from_user. Maybe @moazin , you can make the change and file a pull request? Or we will see if we could find some time to to it in next release (2023.1).

ndcontini, 2023-11-09:
Any chance we can see this fix in an upcoming release? I have experienced this in some of my experiments as well.