Xilinx/XRT #6867: Memory leak in XRT library

epingel2, 2022-07-20:
After noticing unexpected behavior while using the video-sdk, as well as growing memory in my application, It seems that I have identified memory leaks in the xrt library.
Here is the Valgrind output of running the transcode example provided in the video sdk:

```
==15393== Memcheck, a memory error detector
==15393== Copyright (C) 2002-2017, and GNU GPLd, by Julian Seward et al.
==15393== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==15393== 
Frame=  256 Total FPS=63.728 Current FPS=70.810
Frames Transcoded: 300, Time Elapsed: 4.797
Total FPS: 62.540
==15393== 
==15393== HEAP SUMMARY:
==15393==     in use at exit: 90,648 bytes in 118 blocks
==15393==   total heap usage: 425,033 allocs, 424,915 frees, 187,994,255 bytes allocated
==15393== 
==15393== 32 bytes in 1 blocks are definitely lost in loss record 12 of 52
==15393==    at 0x4C3217F: operator new(unsigned long) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==15393==    by 0x4E56412: xma_thread2() (in /opt/xilinx/xrt/lib/libxma2api.so.2.10.50)
==15393==    by 0x51306DE: ??? (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25)
==15393==    by 0x6DF66DA: start_thread (pthread_create.c:463)
==15393==    by 0x608A61E: clone (clone.S:95)
==15393== 
==15393== 32 bytes in 1 blocks are definitely lost in loss record 13 of 52
==15393==    at 0x4C3217F: operator new(unsigned long) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==15393==    by 0x4E54FD8: xma_thread1() (in /opt/xilinx/xrt/lib/libxma2api.so.2.10.50)
==15393==    by 0x51306DE: ??? (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25)
==15393==    by 0x6DF66DA: start_thread (pthread_create.c:463)
==15393==    by 0x608A61E: clone (clone.S:95)
==15393== 
==15393== 716 (512 direct, 204 indirect) bytes in 1 blocks are definitely lost in loss record 42 of 52
==15393==    at 0x4C3217F: operator new(unsigned long) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==15393==    by 0x4E62EB9: xma_scaler_session_create (in /opt/xilinx/xrt/lib/libxma2api.so.2.10.50)
==15393==    by 0x1123B6: xlnx_scal_session (xlnx_scaler.c:346)
==15393==    by 0x11332C: xlnx_tran_session_create (xlnx_transcoder.c:362)
==15393==    by 0x117393: main (xlnx_transcoder_app.c:186)
==15393== 
==15393== 3,580 (2,560 direct, 1,020 indirect) bytes in 5 blocks are definitely lost in loss record 47 of 52
==15393==    at 0x4C3217F: operator new(unsigned long) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==15393==    by 0x4E5BA09: xma_filter_session_create (in /opt/xilinx/xrt/lib/libxma2api.so.2.10.50)
==15393==    by 0x118D87: xlnx_la_create (xlnx_lookahead.c:250)
==15393==    by 0x114779: xlnx_enc_la_init (xlnx_encoder.c:478)
==15393==    by 0x115592: xlnx_enc_session (xlnx_encoder.c:908)
==15393==    by 0x1133F5: xlnx_tran_session_create (xlnx_transcoder.c:375)
==15393==    by 0x117393: main (xlnx_transcoder_app.c:186)
==15393== 
==15393== 3,580 (2,560 direct, 1,020 indirect) bytes in 5 blocks are definitely lost in loss record 48 of 52
==15393==    at 0x4C3217F: operator new(unsigned long) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==15393==    by 0x4E5A6BE: xma_enc_session_create (in /opt/xilinx/xrt/lib/libxma2api.so.2.10.50)
==15393==    by 0x1155EF: xlnx_enc_session (xlnx_encoder.c:921)
==15393==    by 0x1133F5: xlnx_tran_session_create (xlnx_transcoder.c:375)
==15393==    by 0x117393: main (xlnx_transcoder_app.c:186)
==15393== 
==15393== 10,240 bytes in 5 blocks are definitely lost in loss record 50 of 52
==15393==    at 0x4C33B25: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==15393==    by 0x114A06: xlnx_enc_update_props (xlnx_encoder.c:564)
==15393==    by 0x1117B8: xlnx_tran_update_props (xlnx_transcoder_parser.c:310)
==15393==    by 0x111AE0: xlnx_tran_parser (xlnx_transcoder_parser.c:402)
==15393==    by 0x117316: main (xlnx_transcoder_app.c:173)
==15393== 
==15393== 63,573 (512 direct, 63,061 indirect) bytes in 1 blocks are definitely lost in loss record 52 of 52
==15393==    at 0x4C3217F: operator new(unsigned long) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==15393==    by 0x4E58E3E: xma_dec_session_create (in /opt/xilinx/xrt/lib/libxma2api.so.2.10.50)
==15393==    by 0x10D567: xlnx_dec_session (xlnx_decoder.c:473)
==15393==    by 0x1132AC: xlnx_tran_session_create (xlnx_transcoder.c:351)
==15393==    by 0x117393: main (xlnx_transcoder_app.c:186)
==15393== 
==15393== LEAK SUMMARY:
==15393==    definitely lost: 16,448 bytes in 19 blocks
==15393==    indirectly lost: 65,305 bytes in 71 blocks
==15393==      possibly lost: 0 bytes in 0 blocks
==15393==    still reachable: 8,895 bytes in 28 blocks
==15393==         suppressed: 0 bytes in 0 blocks
==15393== Reachable blocks (those to which a pointer was found) are not shown.
==15393== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==15393== 
==15393== For counts of detected and suppressed errors, rerun with: -v
==15393== ERROR SUMMARY: 7 errors from 7 contexts (suppressed: 0 from 0)
```

There seems to be leaks in the destroy functions for each of the sessions. In each session type (decoder/scaler/filter/encoder) there is a `XmaHwSessionPrivate` object being allocated with new and assigned to `session->base.hw_session.private_do_not_use`. This is not destroyed in the destroy function. Doing so seems to cause issues elsewhere.

sarab96, 2022-08-01:
What exactly is the issue you see?
What is your setup and application trying to do?
Growing memory could be due to many reasons?
Usually xclbin cached in memory is biggest memory consumer

sarab96, 2022-08-04:
For good scalability and performance:
1) Service should be able to execute/start multiple processes
2) Each process will execute xma_initialize  once
3) Each process may use multiple devices. Processes may share a device
4) Setup reservation & cu allocation (XRM setup) so that multiple processes can work well

Please consult your Video/XRM support person to get more info on using multiple processes

closed 2022-08-04 by sarab96