Xilinx/XRT #6942: "xdp_hw_emu_device_offload_plugin" does not exist

mahdiaminik, 2022-08-20:
Hi.
I'm very newbie in hardware design system and i'm trying to build `XRT` in docker for `pynq` system also installed via `pip3`, without any Vitis, Vivado or ISE. just for connecting pynq to XRT to access PL from PS.
(according to [https://discuss.pynq.io/t/docker-xilinx-platforms-pynq/1962](https://discuss.pynq.io/t/docker-xilinx-platforms-pynq/1962))
(my board : [http://www.alinx.com/en/index.php/default/content/114.html](http://www.alinx.com/en/index.php/default/content/114.html))
and when i'm tying to build XRT i see :

```
CMake Error at runtime_src/xdp/CMakeLists.txt:406 (add_dependencies):
  The dependency target "xrt_hwemu" of target
  "xdp_hw_emu_device_offload_plugin" does not exist.

CMake Error at runtime_src/xdp/CMakeLists.txt:407 (add_dependencies):
  The dependency target "xrt_hwemu" of target "xdp_hw_emu_pl_deadlock_plugin"
  does not exist.
```

And here is my `Dockerfile` :

```
FROM    ubuntu:20.04
ENV     DEBIAN_FRONTEND="noninteractive" TZ=Asia/Tehran
# for tzdata in xrtdeps.sh dependency tree
RUN     ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && \
        echo $TZ > /etc/timezone && \
        apt-get update && \
        apt-get install -y tzdata git 
WORKDIR /xrt
RUN     git clone https://github.com/Xilinx/XRT.git . && \
        sed -i "s/apt/apt-get/g; s/clang-tidy/-y clang-tidy/g" /xrt/src/runtime_src/tools/scripts/xrtdeps.sh
RUN     /xrt/src/runtime_src/tools/scripts/xrtdeps.sh -docker
RUN     cd /xrt/build && ./build.sh
```

A suggestion :
Please upload the final `.deb` file of installing XRT in Releases section.(specially `armhf` arch)
Thank you.

uday610, 2022-08-25:
@jvillarre , can you see if you have any insight here

mahdiaminik, 2022-09-01:
> The two debug/profile plugins listed do have a dependency on the hardware emulation libraries being built, and the hardware emulation libraries have a restriction where they are only built if the system processor is x86_64. If running build.sh on the Zynq board itself the hardware emulation libraries will not be built.
> 
> XRT can be built on an x86 for the zynq by using the build_edge.sh script.

Thanks for your answer.
But how did they (pynq devs) build [image](https://pynq.readthedocs.io/en/v2.2.1/getting_started/pynq_image.html#supported-boards) within XRT for `armhf` arch? 
via crosscompile???

uday610, 2022-10-19:
@mahdiaminik , we dont know about pync developer, they are different team and they may have their custom flow. You may reach out to them to know about this. From our XRT side, we officially support build_edge.sh script as mentioned in the previous comment

yyx32, 2024-04-01:
Hi @mahdiaminik ,have you solve this error?