Xilinx/XRT #6943: [xbutil] ERROR: stoi when running `xbutil reset`

IskandarZhang, 2022-08-21:
I got 3 Alveo U50 cards on my work station and I want to reset the device after an interrupted programming so I execute the command:
```
xbutil reset --device [0000:03:00.1] : xilinx_u50_gen3x16_xdma_201920_3
```
but it didn't work, and the error message is shown below
```
XRT build version: 2.11.634
Build hash: 5ad5998d67080f00bca5bf15b3838cf35e0a7b26
Build date: 2021-06-08 22:08:45
Git branch: 2021.1
PID: 366222
UID: 1026
[Sun Aug 21 02:57:12 2022 GMT]
HOST: c2
EXE: /opt/xilinx/xrt/bin/unwrapped/xbutil2
[xbutil] ERROR: stoi

```
Thanks for help in advance.

uday610, 2022-08-25:
Can you try `xbutil reset --device 0000:03:00.1` , It seems you are using a longer string than necessary which is why "stoi" is complaining.

IskandarZhang, 2022-08-25:
> Can you try `xbutil reset --device 0000:03:00.1` , It seems you are using a longer string than necessary which is why "stoi" is complaining.

And it worked, thanks alot!

closed 2022-08-25 by IskandarZhang