Xilinx/XRT #6949: error for build xrt runtime

pardon-hnu, 2022-08-22:
When i built the xrt runtime, there were some errors occuring：

```
hnu@hnu-PowerEdge-R740:~/Disk0/ParDon/Xilinx/XRT-202020.2.8.743/build$ sudo ./build.sh
cmake -DRDI_CCACHE=0 -DCMAKE_BUILD_TYPE=Debug -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCMAKE_TOOLCHAIN_FILE= ../../src
-- Host system processor is x86_64
-- Target system processor is x86_64
-- Looking for DRM - found at /usr 2.4.107
-- Looking for OPENCL - found at /usr 2.1 /usr/include
-- Looking for GIT - found at /bin/git
-- Found Boost: /lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: system filesystem 
-- XRT EA eula files  /home/hnu/Disk0/ParDon/Xilinx/XRT-202020.2.8.743/src/../LICENSE
fatal: 不是 git 仓库（或者直至挂载点 /home/hnu 的任何父目录）
停止在文件系统边界（未设置 GIT_DISCOVERY_ACROSS_FILESYSTEM）。
fatal: 不是 git 仓库（或者直至挂载点 /home/hnu 的任何父目录）
停止在文件系统边界（未设置 GIT_DISCOVERY_ACROSS_FILESYSTEM）。
fatal: 不是 git 仓库（或者直至挂载点 /home/hnu 的任何父目录）
停止在文件系统边界（未设置 GIT_DISCOVERY_ACROSS_FILESYSTEM）。
fatal: 不是 git 仓库（或者直至挂载点 /home/hnu 的任何父目录）
停止在文件系统边界（未设置 GIT_DISCOVERY_ACROSS_FILESYSTEM）。
-- Platform/Linux (Ubuntu) (Kernel 5.15.0-46-generic)
-- Compiler: /bin/c++ /bin/cc
-- Found Boost: /lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: system filesystem program_options 
-- Copying test data
-- GTest include dirs: '/usr/include'
-- Configuring CMake to run unit test after building
-- XRT xrt++ header files
-- xrt++.hpp
-- xrtexec.hpp
CMake Warning at runtime_src/ert/CMakeLists.txt:52 (message):
  ****************************************************************

  No firmware files built or copied, resulting XRT package will be missing
  ERT scheduler firmware.  Use build.sh -ertfw <dir> to specify path to a
  directory with firmware to copy during XRT build.

  

  ****************************************************************

-- XRT experimental header files
-- xrt-next.h
-- xrt_aie.h
-- xrt_graph.h
-- xrt_bo.h
-- xrt_device.h
-- xrt_enqueue.h
-- xrt_error.h
-- xrt_ini.h
-- xrt_kernel.h
-- xrt_profile.h
-- xrt_uuid.h
-- xrt_xclbin.h
-- xclbin_util.h
-- xclbin-util.h
-- XRT deprecated header files
-- xrt.h
bfl=Boost::filesystem
-- XRT header files
-- mailbox_proto.h
-- XRT version: 2.8.0
cpackLin XRT_BOOST_VER_STR=1.71.0
-- Debug DEB package
CMake Warning at CMake/lint.cmake:10 (message):
  -- run-clang-tidy not found, static code analysis disabled
Call Stack (most recent call first):
  CMake/nativeLnx.cmake:161 (include)
  CMakeLists.txt:80 (include)

-- XRT DRIVER SRC BASE DIR /home/hnu/Disk0/ParDon/Xilinx/XRT-202020.2.8.743/src/runtime_src/core
[... 203 lines cut ...]
  * The installation package was faulty and references files it does not
  provide.

CMake Error in xma/src/xmaapi/CMakeLists.txt:
  Imported target "Boost::filesystem" includes non-existent path

    "/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

real	0m1.677s
user	0m0.554s
sys	0m0.130s
```
I want to know ​How is this be solved or where can I download xrt_<version>.deb for vitis 2020.2(ubuntu 20.04) directly instead of building it from the source code.

uday610, 2022-08-25:
As far as I know , XRT is still not supported for 5.15.0-46-generic kernel, @chvamshi-xilinx , can provide update

pardon-hnu, 2022-08-25:
> As far as I know , XRT is still not supported for 5.15.0-46-generic kernel, @chvamshi-xilinx , can provide update
Thank you,  I changed the linux kernel version to 5.8.0-63 and built XRT successful, But the installing of XRT failed. 
`hnu@hnu-PowerEdge-R740:~/Disk0/ParDon/Xilinx/XRT/XRT/build/Release$ sudo apt reinstall ./xrt_202020.2.8.0_20.04-amd64-xrt.deb
正在读取软件包列表... 完成
正在分析软件包的依赖关系树       
正在读取状态信息... 完成       
注意，选中 'xrt' 而非 './xrt_202020.2.8.0_20.04-amd64-xrt.deb'
下列软件包是自动安装的并且现在不需要了：
  amd64-microcode intel-microcode iucode-tool thermald
使用'sudo apt autoremove'来卸载它(它们)。
升级了 0 个软件包，新安装了 0 个软件包，重新安装了 1 个软件包，要卸载 0 个软件包，有 1 个软件包未被升级。
解压缩后会消耗 0 B 的额外空间。
获取:1 /home/hnu/Disk0/ParDon/Xilinx/XRT/XRT/build/Release/xrt_202020.2.8.0_20.04-amd64-xrt.deb xrt amd64 2.8.0 [7,958 kB]
(正在读取数据库 ... 系统当前共安装有 485361 个文件和目录。)
准备解压 .../xrt_202020.2.8.0_20.04-amd64-xrt.deb  ...
Unregistering old XRT Linux kernel module sources 2.8.0 from dkms on Ubuntu/Debian

-------- Uninstall Beginning --------
Module:  xrt
Version: 2.8.0
Kernel:  5.8.0-63-generic (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

xocl.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.8.0-63-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

xclmgmt.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.8.0-63-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

DKMS: uninstall completed.

------------------------------
Deleting module version: 2.8.0
completely from the DKMS tree.
------------------------------
Done.
Cleanup is skipped for package upgrade/downgrade/re-install on Ubuntu/Debian
正在解压 xrt (2.8.0) 并覆盖 (2.8.0) ...
正在设置 xrt (2.8.0) ...
Unloading old XRT Linux kernel modules
rmmod: ERROR: Module xclmgmt is not currently loaded
Invoking DKMS common.postinst for xrt
Loading new xrt-2.8.0 DKMS files...
Building for 5.8.0-63-generic 5.15.0-46-generic
Building initial module for 5.8.0-63-generic
Secure Boot not enabled on this system.
Done.

xocl.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.8.0-63-generic/updates/dkms/

xclmgmt.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.8.0-63-generic/updates/dkms/

depmod...

DKMS: install completed.
Module build for kernel 5.15.0-46-generic was skipped since the
kernel headers for this kernel does not seem to be installed.
Finished DKMS common.postinst
Loading new XRT Linux kernel modules
modprobe: ERROR: could not insert 'xclmgmt': Unknown symbol in module, or unknown parameter (see dmesg)
Installing MSD / MPD daemons
W: Repository is broken: xrt:amd64 (= 2.8.0) has no Size information
`
Dmesg information：
`[90636.236173] xclmgmt: Unknown symbol fpga_mgr_create (err -2)
[90636.236276] xclmgmt: Unknown symbol fpga_mgr_unregister (err -2)
[90636.236319] xclmgmt: Unknown symbol fpga_mgr_register (err -2)
[90636.236335] xclmgmt: Unknown symbol fpga_mgr_free (err -2)
[90990.924844] xclmgmt: Unknown symbol fpga_mgr_create (err -2)
[90990.924937] xclmgmt: Unknown symbol fpga_mgr_unregister (err -2)
[90990.924973] xclmgmt: Unknown symbol fpga_mgr_register (err -2)
[90990.924987] xclmgmt: Unknown symbol fpga_mgr_free (err -2)
`

keryell, 2022-08-25:
The latest XRT might not be supported but seems to work up to 5.18 and there is a PR for 5.19 https://github.com/Xilinx/XRT/pull/6908