Xilinx/XRT #7002: Simplify XRT native C++ API usage with less headers to include

keryell, 2022-09-07:
It is cumbersome to include many headers for an XRT application.
There should be a single header for most use cases, like `xrt/xrt.h` or `xrt/xrt.hpp` or `xrt/xrt++.h` or *[bikesheding]*.
I think it makes sense today since XRT is becoming less experimental. There could be other includes for very specific/experimental features or extensions.

bryanloz-xilinx, 2022-09-20:
I agree here with @keryell. I also think its strange that the existing `xrt.h` is included by `xrt/xrt_device.h` it should be the other way around.

stsoe, 2022-09-20:
yeah. well, but that xrt.h is completely different from what is being discussed here.

xrt.h is a legacy header file on top of which XRT C++ APIs are built, for example xcl typedefs are in xrt.h and are used in the C++ APIs.  This was done early on when the C++ APIs were used in conjunction with xcl APIs.  No longer the case, so some cleanup should be done.

Confusing indeed.  I will address it as part of this issue.