Xilinx/XRT #7010: Whether using x86 host to control AIE graph through XRT C++ API is supported for vck5000 prod card ?

zzy82518996, 2022-09-09:
We have a vck5000 development card installed on the server. According to the public available tutorial, I tried to deploy some simple application algorithms onto vck5000. Currently I am able to use the x86 host to control either PL kernels only or PL+AIE kernels. However, looks like the AIE Graph control is done by Cortex-72, which means the X86 host is not able to control the AIE graph through XRT C++ API. I am not sure whether my understanding is correct.  I am not able to find the definition of function such as xrt::graph or xrt::graph::run in xrt/lib, so I guess the implemtation is insdie Vitis/aietools/lib ? What if I only want to use AIE which feeds AIE data from global memory through NOC, not through PL kernel ? Can I use X86 host for AIE graph controlling ? I have tried using XRT C++ API, but got many undefined reference errors:

```
/home/zz546/temp/vadd_aie_only_myself/sw/src/host.cpp:49: undefined reference to `xrt::graph::graph(xrt::device const&, xrt::uuid const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, xrt::graph::access_mode)'
/usr/bin/ld: /home/zz546/temp/vadd_aie_only_myself/sw/src/host.cpp:51: undefined reference to `xrt::graph::reset() const'
/usr/bin/ld: /home/zz546/temp/vadd_aie_only_myself/sw/src/host.cpp:52: undefined reference to `xrt::aie::bo::sync(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, xclBOSyncDirection, unsigned long, unsigned long)'
/usr/bin/ld: /home/zz546/temp/vadd_aie_only_myself/sw/src/host.cpp:53: undefined reference to `xrt::aie::bo::sync(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, xclBOSyncDirection, unsigned long, unsigned long)'
/usr/bin/ld: /home/zz546/temp/vadd_aie_only_myself/sw/src/host.cpp:55: undefined reference to `xrt::graph::run(unsigned int)'
/usr/bin/ld: /home/zz546/temp/vadd_aie_only_myself/sw/src/host.cpp:56: undefined reference to `xrt::graph::end(unsigned long)'
/usr/bin/ld: /home/zz546/temp/vadd_aie_only_myself/sw/src/host.cpp:57: undefined reference to `xrt::aie::bo::sync(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, xclBOSyncDirection, unsigned long, unsigned long)'
/usr/bin/ld: /tmp/ccVyRyJ5.o: in function `xrt::aie::device::device<int>(int&&, xrt::aie::access_mode)':
/opt/xilinx/xrt/include/xrt/xrt_aie.h:75: undefined reference to `xrt::aie::device::open_context(xrt::aie::access_mode)'
collect2: error: ld returned 1 exit status
```
Does this means if I only want AIE kernels to run, only embedded platform such as vck5000 ES1 or vck190 is supported ? Is it also possible for vck5000 prod card which conntect host machine through PCIe ? Thanks.

uday610, 2022-09-13:
No, any VCK5000 card (ES1, Prod, etc) which are publicly available today does not support controlling (using XRT C++ APIs or through any API) AIE from X86... From the X86 host program, you can exercise PL and PL+AIE which you discovered already. 

Thanks

zzy82518996, 2022-09-17:
Got it. Thanks for the reply ! This helps a lot.

closed 2022-09-17 by zzy82518996