Xilinx/XRT #7054: Question about locks in xrt::kernel

byeongkeonLee, 2022-09-19:
Dear,

While I was programming FPGA with xrt, I encountered an issue.

I have two xrt::kernel: kernel_0 and kernel_1.

I launched kernel_0 with kernel_0.run(), then kernel_0.wait().
When kernel_0 is waiting by done, I launched kernel_1.run(), then kernel_1.wait(). 

However, even if kernel_1.wait() is possible to wake up prior to kernel_0.wait(), kernel_1.wait() doesn't awake until kernel_0 awakes.

I have found each xrt::kernel has its own lockguard, but I can not understand kenrel_0 and kernel_1 behavior (It looks like they have a global lock).

Am I correct? 

Sincerely,

uday610, 2022-09-22:
Hello,
It is not sure what exactly your code sequence is, but if you write something like this:

```
kernel_0.run();
kernel_0.wait();
kernel_1.run();
kernel_1.wait();
```
Then obviously the host thread will wait at the line where you did ``kernel_0.wait()`` as that is a blocking statement, ``kernel_1.run()`` will not even start running in that case until kernel_0 finishes. 

If you write a code like this

```
kernel_0.run(); 
kernel_1.run();
kernel_1.wait();
kernel_0.wait();
```

And if you have only 1 kernel in hardware (which is represented by both kernel_0 and kernel_1), then obviously kernel_1 cannot be executed before kernel_0 is finished. 

In short, there can be myriads of situations that you are observing is possible. If both the kernel objects are denoting completely different hardware resources, and you launch them together, then also I think your host code should not assume which one finishes first. Even if they are running in parallel using different hardware resources, then the only thing we can guarantee in host code is that they should both finish. As scheduling and the actual time taken by both the kernels are not very deterministic, so we should not assume this particular one will definitely finish faster (or any similar assumption).

byeongkeonLee, 2022-10-05:
I got it!

Thank you for your detailed answer.

closed 2022-10-05 by byeongkeonLee