Xilinx/XRT #7137: Zocl backward compatibility brocken

xlz-epimor, 2022-10-27:
Hi,

When compiling a xrt/zocl dependent app on petalinux 2022.1 with esdk update3, the resulting application is built with xrt/zocl 2.13.479.

When i'm trying to install this application with its generated rpm on a kv260 starter kit with xrt/zocl 2.14.0 version installed on it, i get an error when trying to open device through xrt c++ API (xrt::device device(0);): unable to open device 0.

When i downgrade the zocl version to 2.13.479 the application success in opening device 0.

It seems that the new version of zocl is not backward compatible.

xlz-epimor, 2022-10-27:
After some research in log install of zocl dependency, it appears that the rpm post install script fails when trying to load the zocl module:

  Installing       : zocl-202220.2.14.0-r0.0.xilinx_k26_kv                                                                                                                                                                              5/12
  Running scriptlet: zocl-202220.2.14.0-r0.0.xilinx_k26_kv                                                                                                                                                                              5/12
%post(zocl-202220.2.14.0-r0.0.xilinx_k26_kv): scriptlet start
%post(zocl-202220.2.14.0-r0.0.xilinx_k26_kv): execv(/bin/sh) pid 2109
+ set -e
+ '[' -n '' ']'
+ echo 'Unloading old XRT Linux kernel modules'
Unloading old XRT Linux kernel modules
+ echo 'Loading new XRT Linux kernel modules'
Loading new XRT Linux kernel modules
+ modprobe zocl
modprobe: FATAL: Module zocl not found in directory /lib/modules/5.15.19-xilinx-v2022.1
%post(zocl-202220.2.14.0-r0.0.xilinx_k26_kv): waitpid(2109) rc 2109 status 100
warning: %post(zocl-202220.2.14.0-r0.0.xilinx_k26_kv) scriptlet failed, exit status 1

Error in POSTIN scriptlet in rpm package zocl

xlz-epimor, 2022-10-27:
Moving this issue in https://github.com/Xilinx/meta-xilinx as it is linked to zocl petalinux rpm post inst script

xlz-epimor, 2022-11-04:
Hi @chvamshi-xilinx 
From my research here is the sources of zocl driver https://github.com/Xilinx/XRT/tree/master/src/runtime_src/core/edge/drm
If it can help, i am working on a kv260 starter kit

chvamshi-xilinx, 2022-11-04:
yes, We own the source code, but my question is about the way zocl is compiled. Based on below error message, it seems zocl is not even present in below path
**modprobe: FATAL: Module zocl not found in directory /lib/modules/5.15.19-xilinx-v2022.1**

xlz-epimor, 2022-11-04:
Here is what i found from my petalinux project:
https://github.com/Xilinx/meta-xilinx/tree/master/meta-xilinx-core/recipes-xrt

chvamshi-xilinx, 2022-11-04:
couple of other questions for more clarity
1) Test is working if you are using 2022.1 Petalinux Image/rootfs and XRT/Zocl also from 2022.1. am i correct?
2) You are just trying to change zocl by installing zocl rpm. It didnt install zocl properly. From where did you get this rpm. Looks like this rpm is not compatible for 2022.1 Petalinux

xlz-epimor, 2022-11-04:
This test is with 2022.1 petalinux image
The rpm is from the xilinx repository https://petalinux.xilinx.com/sswreleases/ and as there is not separation between rpm for 2022.1 and those for 2022.2 it install the newest version (the one for 2022.2)

chvamshi-xilinx, 2022-11-04:
There is a minor kernel version change in Petalinux 2022.2 from 2022.1. Modules wont be compatible between the releases.
You may need to change your dnf repo path to pick the right version.

xlz-epimor, 2022-11-04:
Shouldn't be an update of the Xilinx repo with separate directory for 2022.1 and 2022.2 rpms ? like it was made for 2021.1 and 2021.2 ? Furthermore if rpms for 2022.2 are not compatible with 2022.1

chvamshi-xilinx, 2022-11-04:
I am not expert in that area, I have to check with Yocto team..