Xilinx/XRT #7140: OpenCV cascade detector causes Segmentation fault on KR260

Satoski-H, 2022-10-28:
Hi all,
I was trying OpenCV cascade detector on KR260 and I got Segmentation fault.
My environment is as follows
Board : KR260
OS : Xilinx certified Ubuntu 22.04.01 LTS.
OpenCV version : 4.5.4+dfsg-9ubuntu4
g++ version : 11.3.0

When I ran my program on GDB, I got this message.
-------
 Segmentation fault.
0x0000ffffec549120 in xrt_core::get_os_info(boost::property_tree::basic_ptree<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&) () from /lib/libxrt_coreutil.so.2
-------
It seems that XRT is running, even though I don't use any PL logic.
So, my questions are : 
1. Why is XRT used on just OpenCv program?
2. Is the Segmentation fault I got due to XRT?

I attach my program and GDB log file.
[opencv_cascade.zip](https://github.com/Xilinx/XRT/files/9885865/opencv_cascade.zip)

Any help would be greatly appreciated.
Best regards,

chvamshi-xilinx, 2022-11-04:
Hi @Satoski-H , Can you share the full back trace of crash

Satoski-H, 2022-11-07:
Hi @chvamshi-xilinx,
Thank you for your reply. I attached GDB log and its screenshot.
Best regards,
[log.txt](https://github.com/Xilinx/XRT/files/9948224/log.txt)
[image]