Xilinx/XRT #7179: Build stops because python_binding test fails (minor)

fab7, 2022-11-18:
Hi, 
FYI, the XRT build stops because the **Test 3: python_binding** fails. 

To get a successful build, I need to skip the unit tests with `./build.sh -noctest`.

```
Test project /home/fab/Downloads/Xilinx/GitHub-XRT/build/Debug
      Start  1: xbutil2
 1/17 Test  #1: xbutil2 ...........................   Passed    0.09 sec
      Start  2: xbmgmt2
 2/17 Test  #2: xbmgmt2 ...........................   Passed    0.08 sec
      Start  3: python_binding
 3/17 Test  #3: python_binding ....................***Failed    0.08 sec
Traceback (most recent call last):
  File "/home/fab/Downloads/Xilinx/GitHub-XRT/src/../tests/python/200_binding/200_main.py", line 18, in <module>
    import pyxrt as pp
ModuleNotFoundError: No module named 'pyxrt'
```

The build is executed in the following conditions:
- XRT version `4f1341eb6fe407356d0322598956d8e1f596f72c`
- RHEL8.6 `4.18.0-372.19.1 x86_64`

qianyich, 2023-01-23:
I am having the same issue. Have you figured it out?

fab7, 2023-02-06:
[ Solved ] After I installed _Python 3.9_ (using `sudo dnf install python39`) and _pybind_ (using `sudo pip3.9 install pybind11`).

closed 2023-02-06 by fab7