Xilinx/XRT #7180: RHEL 8.7 error: conflicting types for ‘xocl_gem_prime_vmap’

fab7, 2022-11-18:
Hi XRT community,
I know that RHEL8.7 is not (yet) officially supported but I needed to upgrade one of our servers to this very last release of RHEL8 that came out on November 9th, 2022. 
This may require yet another pragma directive in the XOCL driver and I would really appreciate if you could consider this issue. Thanks in advance.

The system and XRT configurations are:
- XRT version  `2.15.0` 
- Git hash `4f1341eb6fe407356d0322598956d8e1f596f72c`
- RHEL8.7  `4.18.0-425.3.1.el8.x86_64`
- XRT

```
/var/lib/dkms/xrt/2.15.0/build/driver/xocl/userpf/xocl_bo.c:1291:5: error: conflicting types for ‘xocl_gem_prime_vmap’
 int xocl_gem_prime_vmap(struct drm_gem_object *obj, struct XOCL_MAP_TYPE *map)
```

```
#if LINUX_VERSION_CODE < KERNEL_VERSION(5, 11, 0) && !defined(RHEL_8_5_GE)
void *xocl_gem_prime_vmap(struct drm_gem_object *obj)
{
        struct drm_xocl_bo *xobj = to_xocl_bo(obj);

        BO_ENTER("xobj %p", xobj);
        return xobj->vmapping;
}

void xocl_gem_prime_vunmap(struct drm_gem_object *obj, void *vaddr)
{

}
#else
int xocl_gem_prime_vmap(struct drm_gem_object *obj, struct XOCL_MAP_TYPE *map)
{
        struct drm_xocl_bo *xobj = to_xocl_bo(obj);

        BO_ENTER("xobj %p", xobj);
        XOCL_MAP_SET_VADDR(map, xobj->vmapping);

        return 0;
}

void xocl_gem_prime_vunmap(struct drm_gem_object *obj, struct XOCL_MAP_TYPE *map)
{

}
#endif
```

keryell, 2022-11-21:
Can you send a PR?

chvamshi-xilinx, 2022-12-01:
Hi @fab7 , RHEL 8.7 Support is not enabled in XRT yet. We are supporting RHEL 9.0 in this release. We will fix this issue as part of that support.

keryell, 2022-12-01:
I guess if it works for RHEL 9.0 it should work for 8.7 too, perhaps with minor preprocessor hack in some files?

fab7, 2023-02-06:
[ Solved ] It seems that issue #7241 fixed the same issue.

closed 2023-02-06 by fab7