Xilinx/XRT #7471: Add `xbutil configure --host-mem` status to `xbutil examine`

mortbopet, 2023-03-27:
Currently, there is no way of knowing, whether `host-mem` has been enabled on a device. This would be very convenient to be able to query, seeing as enabling host-mem requires sudo privileges, and all applications that leverage host-mem might not have the privileges to enable it. Thus, as a way to gracefully inform about a potential issue, it would be nice if `xbutil examine` provided information (in stdout and json) about:

1. Is host-mem enabled for the given device?
2. If so, what is the current size of the host-mem buffer?

P.S.: I think this goes for everything that can be enabled via. `xbutil configure` (e.g. p2p) and their configuration parameters.

uday610, 2023-03-29:
Did you try `xbutil examine --device <bdf> -r memory` ? It should show how much host memory is enabled.

mortbopet, 2023-04-03:
To clarify, I was trying to say the **single** `xbutil examine` command dump **every possible piece** of information available would be very helpful. This would allow us to query the JSON dynamically for any possible piece of information that we'd like to know, instead of having a bunch of separate calls to `xbutil` to dump individual pieces of JSON information.

When i'm dumping something to JSON, i'm not particularly concerned about things being cluttered/optimized for human readability - i'd like to have something that is optimized for analyzability/queryability.

uday610, 2023-04-03:
Yes, you can dump everything in JSON format by a single xbutil command too. 
xbutil examine --device <bdf> --report  all --format JSON --output all_dump.json

mortbopet, 2023-04-03:
I rest my case :) thank you!