Xilinx/XRT #7504: Problem in Build Instructions

hocken-li, 2023-04-15:
In the  Instructions web, there is two part:  1.Building the XRT Software Stack 2. XRT Installation
I run in ubuntu22.04
I finished Step 1. Building the XRT Software Stack
In Step 2, sudo apt install xrt_<version>.deb,  I find so much .deb in ./build/Release, which .deb should I choose??
or I need to download?????

maxzhen, 2023-04-15:
The one named "-xrt" should be installed :-). E.g., XRT/build/Release/xrt_202310.2.15.0_22.04-amd64**-xrt**.deb

hocken-li, 2023-04-18:
@maxzhen Thank you so much!!!!! 
There is one more question: 
Why the version seems to be 202310.2?
The target version I want to build is 2022.2, is there anything wrong?
The next part I want to build is _xrt_202220.2.14.354_22.04-amd64-xrt.deb_  , but I already install rt_202310.2.15.0_22.04-amd64**-xrt**.deb

hocken-li, 2023-04-18:
@maxzhen Thank you so much!!!!!
There is one more question:
Why the version seems to be 202310.2?
The target version I want to build is 2022.2, is there anything wrong?
The next part I want to build is _xilinx-u250-gen3x16-xdma_2022.2_2022_1015_0317-all.deb.tar_  , but I already install rt_202310.2.15.0_22.04-amd64**-xrt**.deb

maxzhen, 2023-04-18:
if you cloned master branch, it is always the latest and greatest xrt. you can try to clone the 2022.2 branch for your specific needs.
but, 2023.1 should also work with 2022.2 shell pkg of u250.