Xilinx/XRT #7518: Problem in install(xocl, xclmgmt andERT firmware)

hocken-li, 2023-04-23:
I run in ubuntu22.04: Follow the instruction in https://xilinx.github.io/XRT/2022.2/html/build.html
When : sudo apt install ./xrt_202220.2.14.0_22.04-amd64-xrt.deb 

There is some problem in xocl and xclmgmt, and also I see "Secure Boot not enabled on this system."
and ERT firmware seems to be skipped, I can't find it in /lib/firmware/xilinx.

Could you tell me is my installation successful? How can I solve these problems?

The follow result is:
`Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'xrt' instead of './xrt_202220.2.14.0_22.04-amd64-xrt.deb'
The following NEW packages will be installed:
  xrt
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/23.6 MB of archives.
After this operation, 104 MB of additional disk space will be used.
Get:1 /home/lab/tools/XRT-2022.2/build/Release/xrt_202220.2.14.0_22.04-amd64-xrt.deb xrt amd64 2.14.0 [23.6 MB]
Selecting previously unselected package xrt.
(Reading database ... 214520 files and directories currently installed.)
Preparing to unpack .../xrt_202220.2.14.0_22.04-amd64-xrt.deb ...
Unpacking xrt (2.14.0) ...
Setting up xrt (2.14.0) ...
Unloading old XRT Linux kernel modules
**_rmmod: ERROR: Module xocl is not currently loaded
rmmod: ERROR: Module xclmgmt is not currently loaded_**
Invoking DKMS common.postinst for xrt
Loading new xrt-2.14.0 DKMS files...
Building for 5.19.0-40-generic
Building initial module for 5.19.0-40-generic
Can't load /var/lib/shim-signed/mok/.rnd into RNG
80CBFDBA2C7F0000:error:12000079:random number generator:RAND_load_file:Cannot op
en file:../crypto/rand/randfile.c:106:Filename=/var/lib/shim-signed/mok/.rnd
..+...+..+.+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*..+
.....+....+...+..............+.+.....+...............+.............+............
+..+...+...+.+...+.....+.............+.....+.......+..+...+............+....+..+
...+....+..+....+...........+....+.........+...+.....+.........+...+.+......+...
..+...+......+....++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
+++*.......+......+......+....++++++++++++++++++++++++++++++++++++++++++++++++++
+++++++++++++++
......+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*........
...+..........+...++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
+++*...+...+........+......+.........+.......+..+......+......+......+...+......
....+.........+...+..+...+....+.....+......+...+.........+..........+..+.+......
..+.+.........+..+.........+.+........+............+.......+..+.+..+.+..+.......
.....+.....................+.......+..+.+...........+......+...+.........+......
.+...+...........+.......+.....+............+..........+...+..+......+.+........
.+......+......+...+..+.......+...........+.+......+.....+.........+....+.......
..+..+......+......................+..+.+........+.......+..+.......+........+..
....+.......+...+...........+.+...+..+....+...+..+...+......+.........+.+.......
..............+........+...+.........+.+............+..+....+.....+......+.+..+.
..+............+......+.+...........+...+....+..+...+......+..........+.....+...
...+...............+....+...........+...+............+...+.......+.........+..+.
...+...+..+...+......+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++++++
-----
**_Secure Boot not enabled on this system._**
Done.

xocl.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.19.0-40-generic/updates/dkms/

xclmgmt.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.19.0-40-generic/updates/dkms/

depmod....
Finished DKMS common.postinst
Loading new XRT Linux kernel modules
Installing MSD / MPD daemons
| Components                   |      Status        |
|------------------------------|--------------------|
| XOCL & XCLMGMT Kernel Driver | Success            |
| XRT USERSPACE                | Success            |
| MPD/MSD                      | Success            |
`

lsyLearn, 2023-05-22:
I also had the same problem, did you solve it?

hocken-li, 2023-05-22:
@lsyLearn It seems to be kernel problem. 
Unsupported kernel 5.19.0
The kernel that is support with XRT 2022.2 with Ubuntu 22.04 is 5.15.0-41-generic.