Xilinx/XRT #7546: XRT ./build.sh failed on 11 - [xclbinutil]:ps-kernel (Failed)

pengjintao, 2023-05-11:
System: linux 5.19.0-41-generic. Ubuntu 22.04.2 LTS, GCC 11.3.0, Xilinx U280.
XRT Build Version: 2.16.0 (master)

error info:

Test project /home/pjt/install/XRT/build/Debug
      Start  1: xbutil2
 1/18 Test  #1: xbutil2 ...........................   Passed    0.00 sec
      Start  2: xbmgmt2
 2/18 Test  #2: xbmgmt2 ...........................   Passed    0.00 sec
      Start  3: python_binding
 3/18 Test  #3: python_binding ....................   Passed    0.01 sec
      Start  4: [xclbinutil]:signing-xclbin_CER
 4/18 Test  #4: [xclbinutil]:signing-xclbin_CER ...   Passed    0.08 sec
      Start  5: [xclbinutil]:signing-xclbin_DER
 5/18 Test  #5: [xclbinutil]:signing-xclbin_DER ...   Passed    0.09 sec
      Start  6: [xclbinutil]:smartnic-syntax
 6/18 Test  #6: [xclbinutil]:smartnic-syntax ......   Passed    0.12 sec
      Start  7: [xclbinutil]:smartnic-format
 7/18 Test  #7: [xclbinutil]:smartnic-format ......   Passed    0.07 sec
      Start  8: [xclbinutil]:soft-kernel
 8/18 Test  #8: [xclbinutil]:soft-kernel ..........   Passed    0.06 sec
      Start  9: [xclbinutil]:partition_metadata
 9/18 Test  #9: [xclbinutil]:partition_metadata ...   Passed    0.05 sec
      Start 10: [xclbinutil]:fixed-kernel
10/18 Test #10: [xclbinutil]:fixed-kernel .........   Passed    0.09 sec
      Start 11: [xclbinutil]:ps-kernel
11/18 Test #11: [xclbinutil]:ps-kernel ............***Failed    0.06 sec

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2a) Read in a PS kernel, updated and validate the sections
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
xclbinutil --input working.xclbin --add-pskernel pskernel.so --dump-section EMBEDDED_METADATA:RAW:embedded_metadata_updated.xml --dump-section IP_LAYOUT:JSON:ip_layout_updated.json --dump-section CONNECTIVITY:JSON:connectivity_updated.json --dump-section MEM_TOPOLOGY:JSON:mem_topology_updated.json --output pskernel_output.xclbin --force
XRT Build Version: 2.16.0 (master)
       Build Date: 2023-05-11 12:00:32
          Hash ID: 073c963d447065b2ef3df1817352a795370a7b55
Reading xclbin file into memory.  File: working.xclbin

Error: No kernels found in the shared library: 'pskernel.so'

Exception('Operation failed with the return code: 255')
Test Status: FAILED
>>> End of unit test <<<
 
========================================================================
Test Directory    : /home/pjt/install/XRT/build/Debug/unit-tests/xclbinutil/ps-kernel
Log File          : runme.log
Error Return Code : 1
Test Status       : FAILED
========================================================================

94% tests passed, 1 tests failed out of 18

Total Test time (real) =   1.10 sec

The following tests FAILED:
	 11 - [xclbinutil]:ps-kernel (Failed)
Errors while running CTest

sunying2024, 2024-04-12:
I have also encountered this problem. Have you found a solution? Thanks a lot

Kylin-2024, 2024-04-12:
I have also encountered this problem. Have you found a solution? Thanks a lot.

hubertzhong, 2024-04-15:
If your server language is set to Chinese, just change it to English and try again :)