Xilinx/XRT #7562: How to start mpd.service

lsyLearn, 2023-05-19:
I installed the XRT on ubuntu20.04
when I run `/opt/xilinx/xrt/bin/xbmgmt examine`， I can get
`XRT
  Version              : 2.14.354
  Branch               : 2022.2
  Hash                 : 43926231f7183688add2dccfd391b36a1f000bea
  Hash Date            : 2022-10-08 09:49:58
  XOCL                 : unknown, unknown
  XCLMGMT              : unknown, unknown
WARNING: xclmgmt version is unknown. Is xclmgmt driver loaded? Or is MSD/MPD running?

Devices present
  0 devices found
BDF  :  Shell  Platform UUID  Device ID  Device Ready*
--------------------------------------------------------

* Devices that are not ready will have reduced functionality when using XRT tools
`
It says I need to start the mpd service，when  I run `systemctl status `, I can get
`● mpd.service - Xilinx Management Proxy Daemon (MPD)
     Loaded: loaded (/etc/systemd/system/mpd.service; enabled; vendor preset: enabled)
     Active: inactive (dead)
  Condition: start condition failed at Fri 2023-05-19 14:31:34 CST; 51min ago

May 19 14:31:13 lenovo systemd[1]: Condition check resulted in Xilinx Management Proxy Daemon (MPD) being skipped.
`
so, I run `cat  /etc/systemd/system/mpd.service` :
`[Unit]
Description=Xilinx Management Proxy Daemon (MPD)
Wants=network-online.target
After=network-online.target
ConditionDirectoryNotEmpty=/dev/xfpga

[Service]
Type=simple
ExecStart=/opt/xilinx/xrt/bin/mpd
Restart=on-failure

[Install]
WantedBy=multi-user.target
`
I find that /dev/xfpga needs to exist in my environment，but I don't have the relevant files in my environment
`ls /dev/xfpga
ls: cannot access '/dev/xfpga': No such file or directory`
So, how do I get the mpd service to start.
run `ls /dev/ | grep xdma`
`xdma0_c2h_0
xdma0_c2h_1
xdma0_c2h_2
xdma0_c2h_3
xdma0_control
xdma0_events_0
...`
run `lspci | grep Xilinx`
`06:00.0 Serial controller: Xilinx Corporation Device 903f`

lsyLearn, 2023-05-22:
when I run :` lspci -vd 10ee:`

> 06:00.0 Serial controller: Xilinx Corporation Device 903f (prog-if 01 [16450])
>         Subsystem: Xilinx Corporation Device 0007
>         Physical Slot: 3
>         Flags: fast devsel, NUMA node 0
>         Memory at d9000000 (64-bit, non-prefetchable) [size=8M]
>         Memory at d9800000 (32-bit, non-prefetchable) [size=64K]
>         Capabilities: [40] Power Management version 3
>         Capabilities: [60] MSI-X: Enable- Count=32 Masked-
>         Capabilities: [70] Express Endpoint, MSI 00
>         Capabilities: [100] Advanced Error Reporting
>         Capabilities: [1c0] Secondary PCI Express
>         Capabilities: [1f0] Virtual Channel

there is no 'Kernel driver in use'
but :` lsmod | grep xclmgmt`

> xclmgmt              1105920  0
> libcrc32c              16384  8 nf_conntrack,nf_nat,xclmgmt,xocl,btrfs,nf_tables,raid456,ip_vs

uday610, 2023-06-07:
Can you please check your OS and kernel version that is officially supported? Maybe you can install the XRT package again by --reinstall and capture and screen log to ensure there is no error.