Xilinx/XRT #7570: Parse xbtop/Record FPGA usage

phip123, 2023-06-01:
Hi,

is there any tool/library that provides the output of xbtop in a machine readable format?
I sadly have troubles with the "build with docker" script, otherwise I would simply change the source code.

I figure there must be some sort of monitoring support that is cloud-native (or similar).

Thanks!

dbenusov-xilinx, 2023-06-07:
Hi @phip123,

We have a [script](https://xilinx.github.io/XRT/master/html/nagios_plugin.html) compatible with [Nagios](https://www.nagios.org/) queries. The output of the nagios script is JSON plus some Nagios flavor text. Running `xbutil examine -d <BDF> -r <Report names> -o <Output file>` will create a JSON with the specified reports.

If using Nagios is out of the question you can invoke `xbutil` directly.

`xbtop` provides the electrical, memory, and dynamic region reports. You could run `xbutil examine -d <BDF> -r memory electrical dynamic-regions -o <Output file>` and send the output JSON to where it is needed.

Is this what you are looking for? We are always looking on ways to improve!

phip123, 2023-06-08:
Hi,

thanks for the comment. Sadly, I'm going to stick with Prometheus and I'm currently working on an exporter that uses xbutil2.

I'm using the xbutil command you suggested - is there any way to direct the JSON output to stdout instead of a file?
Because currently it seems there is no other way than letting xbutil write the JSON file to disk and then read the JSON file, which causes quite some overhead...

Thanks!

dbenusov-xilinx, 2023-06-14:
Hi @phip123,

No worries. Thanks for mentioning Prometheus, looks interesting. 

Unfortunately we cannot. The Nagios plugin functions by creating a [temporary file](https://github.com/dbenusov-xilinx/XRT/blob/3554c144a3021666d38425c87b39c0467f709f6b/src/runtime_src/core/tools/nagios/nagios_plugin.sh#L9) and then deleting it. We can get very close by running `xbutil examine -d <BDF> -r memory electrical dynamic-regions -o /dev/stdout --force`, but, it includes the human readable report which is unfortunate. I'll see if I can add an option to silence the human readable output. Seems like it could be useful.

phip123, 2023-06-14:
Hi @dbenusov-xilinx,

then I'm following the same approach as Nagios, by creating and deleting a temporary file. 
Thanks, I think that would be nice!

Another, somewhat related question, is it possible that the xbutil command locks some resources? For example on instances with multiple FPGAs installed it seems there is some locking going on and even executing the xbutil command in parallel takes longer than individually.

dbenusov-xilinx, 2024-02-06:
I had some time to finally talk to the team about adding the silence feature and it seems like it is more trouble that it is worth considering the uses. If you really need that feature, please open another issue and I will push it further. Thank you for your feedback!

closed 2024-02-06 by dbenusov-xilinx