Xilinx/XRT #7626: Installing xrt on CentOS 9

devschapov, 2023-07-14:
Hi, I have the issue installing xrt_202310.2.15.225_9.0-x86_64-xrt.rpm on
CentOS Stream release 9 (5.14.0-333.el9.x86_64)
with errors while compiling.
Cut from /var/lib/dkms/xrt/2.15.225/build/make.log
DKMS make.log for xrt-2.15.225 for kernel 5.14.0-333.el9.x86_64 (x86_64)
  CC [M]  /var/lib/dkms/xrt/2.15.225/build/driver/xocl/mgmtpf/../subdev/ulite.o
/var/lib/dkms/xrt/2.15.225/build/driver/xocl/mgmtpf/../subdev/ulite.c:469:27: error: initialization of ‘void (*)(struct uart_port *, struct ktermios *, const struct ktermios *)’ from incompatible pointer type ‘void (*)(struct uart_port *, struct ktermios *, struct ktermios *)’ [-Werror=incompatible-pointer-types]
  469 |         .set_termios    = ulite_set_termios,
      |                           ^~~~~~~~~~~~~~~~~
/var/lib/dkms/xrt/2.15.225/build/driver/xocl/mgmtpf/../subdev/ulite.c:469:27: note: (near initialization for ‘ulite_ops.set_termios’)
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:298: /var/lib/dkms/xrt/2.15.225/build/driver/xocl/mgmtpf/../subdev/ulite.o] Error 1

How can it be solved or investigated further to find a cause ?
Thanks.

devschapov, 2023-07-17:
// https://github.com/search?q=repo%3AXilinx%2FXRT%20ulite_set_termios&type=code#:~:text=src/runtime_src/core/pcie/driver/linux/xocl/subdev/ulite.c
  
#if (LINUX_VERSION_CODE > KERNEL_VERSION(6, 1, 0)) || defined(RHEL_9_2_GE)
static void ulite_set_termios(struct uart_port *port, struct ktermios *termios,
			      const struct ktermios *old)
#else
static void ulite_set_termios(struct uart_port *port, struct ktermios *termios,
			      struct ktermios *old)
#endif
// 

OS release/build specific information
# cat /etc/*release*
CentOS Stream release 9
NAME="CentOS Stream"
VERSION="9"
ID="centos"
ID_LIKE="rhel fedora"
VERSION_ID="9"
PLATFORM_ID="platform:el9"
PRETTY_NAME="CentOS Stream 9"
ANSI_COLOR="0;31"
LOGO="fedora-logo-icon"
CPE_NAME="cpe:/o:centos:centos:9"
HOME_URL="https://centos.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_SUPPORT_PRODUCT="Red Hat Enterprise Linux 9"
REDHAT_SUPPORT_PRODUCT_VERSION="CentOS Stream"
CentOS Stream release 9
CentOS Stream release 9
cpe:/o:centos:centos:9

uday610, 2023-07-19:
Please for installation issues reach out to AMD/Xilinx Alveo/XRT Support forum or channel. The OS you are using "CentOS Stream" is not officially supported. Please review tested operating systems: https://support.xilinx.com/s/article/000035189?language=en_US

devschapov, 2023-07-20:
> Please for installation issues reach out to AMD/Xilinx Alveo/XRT Support forum or channel. The OS you are using "CentOS Stream" is not officially supported. Please review tested operating systems: https://support.xilinx.com/s/article/000035189?language=en_US

Hi, thanks for reply.
Already solved with correction ulite.c, void ulite_set_termios().