Xilinx/XRT #7691: xclmgmt seems to failed when v2.12 and later

takasehideki, 2023-09-06:
## summary: expected behavior we wish to do

Hi, all!
We want to use Alveo U50 with Vitis 2022.2 on the Ubuntu 20.04 server machine. However, we cannot observe the connection to the board with XRT v2.12 and later (downloaded from AMD/Xilinx website). Strangely, we can connect to the U50 board with XRT v2.11 (built from source). But it cannot be compatible with `xilinx-u50-gen3x16-xdma_2022.2`, and we cannot install it by deb package.
Can anyone help us by giving the useful information and/or your knowledge?

## our environment

```
$ uname -a 
Linux shugakuin 5.4.0-162-generic #179-Ubuntu SMP Mon Aug 14 08:51:31 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04.6 LTS"

# operated for Alveo U50 with XRT 2.12
$ sudo lspci -vd 10ee:
17:00.0 Processing accelerators: Xilinx Corporation Device d020
	Subsystem: Xilinx Corporation Device 000e
	Flags: fast devsel, IRQ 63, NUMA node 0
	Memory at 39fffc000000 (64-bit, prefetchable) [size=32M]
	Memory at 39fffe000000 (64-bit, prefetchable) [size=128K]
	Capabilities: [40] Power Management version 3
	Capabilities: [48] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [70] Express Endpoint, MSI 00
	Kernel modules: xclmgmt
```

## description

We downloaded packages regarding v2022.2 and Ubuntu 20.4 for Alveo U50 from [AMD/Xilinx website](https://xilinx.com/products/boards-and-kits/alveo/u50.html#gettingStarted), and installed them according to the instruction UG1370.

```
$ sudo apt install ./xrt_202220.2.14.354_20.04-amd64-xrt.deb ./xilinx-*u50*.deb  
<snipped.>
DKMS: install completed.
Finished DKMS common.postinst
Loading new XRT Linux kernel modules
Installing MSD / MPD daemons
| Components                   |      Status        |
|------------------------------|--------------------|
| XOCL & XCLMGMT Kernel Driver | Success            |
| XRT USERSPACE                | Success            |
| MPD/MSD                      | Success            |
Setting up xilinx-u50-gen3x16-xdma-validate (5-3499627) ...
Setting up xilinx-sc-fw-u50 (5.2.20-1.6d4a0da) ...
Setting up xilinx-u50-gen3x16-xdma-5-202210-1-dev (1-3499627) ...
Setting up xilinx-cmc-u50 (1.0.40-3398385) ...
Setting up xilinx-u50-gen3x16-xdma-base (5-3499627) ...
Partition package installed successfully.
Flash the card by following the instructions in the card's installation guide.
```

U50 seems to be detected via PCIe slot. 

```
$ sudo lspci -vd 10ee:
17:00.0 Processing accelerators: Xilinx Corporation Device d020
	Subsystem: Xilinx Corporation Device 000e
	Flags: fast devsel, IRQ 63, NUMA node 0
	Memory at 39fffc000000 (64-bit, prefetchable) [size=32M]
	Memory at 39fffe000000 (64-bit, prefetchable) [size=128K]
	Capabilities: [40] Power Management version 3
	Capabilities: [48] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [70] Express Endpoint, MSI 00
	Kernel modules: xclmgmt

```

But the board cannot be found by XRT v2.14 (`xbmgmt examine`).

```
$ sudo /opt/xilinx/xrt/bin/xbmgmt examine
System Configuration
  OS Name              : Linux
  Release              : 5.4.0-162-generic
  Version              : #179-Ubuntu SMP Mon Aug 14 08:51:31 UTC 2023
  Machine              : x86_64
  CPU Cores            : 20
  Memory               : 257456 MB
  Distribution         : Ubuntu 20.04.6 LTS
  GLIBC                : 2.31
  Model                : Super Server

XRT
  Version              : 2.14.354
  Branch               : 2022.2
  Hash                 : 43926231f7183688add2dccfd391b36a1f000bea
  Hash Date            : 2022-10-08 09:49:58
  XOCL                 : 2.14.354, 43926231f7183688add2dccfd391b36a1f000bea
  XCLMGMT              : 2.14.354, 43926231f7183688add2dccfd391b36a1f000bea

Devices present
  0 devices found
BDF  :  Shell  Platform UUID  Device ID  Device Ready*  
--------------------------------------------------------

* Devices that are not ready will have reduced functionality when using XRT tools

$ sudo /opt/xilinx/xrt/bin/xbmgmt examine --device 17:00.0
ERROR: No devices found
```

When we checked `dmesg`, following message has occurred.

```
[74652.256083] xclmgmt init()
[74652.256873] xclmgmt 0000:17:00.0:  ffff8ee5f0cbb0b0 xclmgmt_probe: Driver: 2.14.354,43926231f7183688add2dccfd391b36a1f000bea
[74652.256883] xclmgmt 0000:17:00.0:  ffff8ee5f0cbb0b0 xclmgmt_probe: probe(pdev = 0x0000000018a00d13, pci_id = 0x0000000043c994a9)
[74652.256889] xclmgmt 0000:17:00.0:  ffff8ee5f0cbb0b0 xclmgmt_probe: ext config space is not accessible, 256
[74652.256904] xclmgmt: probe of 0000:17:00.0 failed with error -22
```

This situation also happened regarding v2022.1 and v2023.1.

In addition, we have tried the XRT deb package by building from this GitHub repository's code.
The tag [202120.2.12.427](https://github.com/Xilinx/XRT/releases/tag/202120.2.12.427) and later also failed in the same situation. Its XRT version is `2.12`.

--- 

Strangely, we can connect to the U50 board with XRT v2.11 (built from source).
The tag is [202110.2.11.680](https://github.com/Xilinx/XRT/releases/tag/202110.2.11.680).

```
$ cd XRT
$ git checkout 202110.2.11.680
$ cd ./src/runtime_src/tools/scripts/xrtdeps.sh
$ cd build
$ ./build.sh -opt
$ cd Release
$ make package
$ sudo apt install ./xrt_202110.2.11.0_20.04-amd64-xrt.deb 
<snipped.>
depmod...

DKMS: install completed.
Finished DKMS common.postinst
Loading new XRT Linux kernel modules
Installing MSD / MPD daemons
```

```
$ sudo lspci -vd 10ee:
17:00.0 Processing accelerators: Xilinx Corporation Device d020
	Subsystem: Xilinx Corporation Device 000e
	Flags: bus master, fast devsel, latency 0, IRQ 63, NUMA node 0
	Memory at 39fffc000000 (64-bit, prefetchable) [size=32M]
	Memory at 39fffe000000 (64-bit, prefetchable) [size=128K]
	Capabilities: [40] Power Management version 3
	Capabilities: [48] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [70] Express Endpoint, MSI 00
	Kernel driver in use: xclmgmt
	Kernel modules: xclmgmt

$ sudo /opt/xilinx/xrt/bin/xbmgmt examine 
System Configuration
  OS Name              : Linux
  Release              : 5.4.0-162-generic
  Version              : #179-Ubuntu SMP Mon Aug 14 08:51:31 UTC 2023
  Machine              : x86_64
  CPU Cores            : 20
  Memory               : 257456 MB
  Distribution         : Ubuntu 20.04.6 LTS
  GLIBC                : 2.31
  Model                : Super Server

XRT
  Version              : 2.11.0
  Branch               : HEAD
  Hash                 : 47ba9cb43c1dd7fcdf742837caaf4a32b0170862
  Hash Date            : 2023-09-06 13:00:54
  XOCL                 : 2.11.0, 47ba9cb43c1dd7fcdf742837caaf4a32b0170862
  XCLMGMT              : 2.11.0, 47ba9cb43c1dd7fcdf742837caaf4a32b0170862

Devices present
  [0000:17:00.0] : xilinx_u50_GOLDEN_9 NOTE: Device not ready for use

$ sudo /opt/xilinx/xrt/bin/xbmgmt examine --device 17:00.0

---------------------------------------
1/1 [0000:17:00.0] : xilinx_u50_GOLDEN
---------------------------------------
Flash properties
  Type                 : spi
  Serial Number        : N/A

Device properties
  Type                 : u50
  Name                 : N/A
  Config Mode          : 3326048144
  Max Power            : N/A

Flashable partitions running on FPGA
  Platform             : xilinx_u50_GOLDEN_9
  SC Version           : INACTIVE
  Platform ID          : N/A

Flashable partitions installed in system
  <none found>        

WARNING  : No shell is installed on the system.

```

But it cannot be compatible with `xilinx-u50-gen3x16-xdma_2022.2`, and we cannot install it by deb package.

```
$ sudo apt install ./xilinx-u50-gen3x16-xdma-5-202210-1-dev_1-3499627_all.deb 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'xilinx-u50-gen3x16-xdma-5-202210-1-dev' instead of './xilinx-u50-gen3x16-xdma-5-202210-1-dev_1-3499627_all.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 xilinx-u50-gen3x16-xdma-5-202210-1-dev : Depends: xrt (>= 2.13.0) but 2.11.0 is to be installed
E: Unable to correct problems, you have held broken packages.
```

Can anyone help us by giving the useful information and/or your knowledge?

uday610, 2023-09-20:
Can you please take this query to Alveo Community Forum?

takasehideki, 2023-09-24:
@uday610 thank you so much for your reply. I also posted the summary of this post (due to character limitation,,, ) and link to here on Alveo Community Forum. 
https://support.xilinx.com/s/question/0D54U00007QyG3tSAF/xclmgmt-seems-to-failed-only-when-using-v212-and-later

closed 2023-10-05 by uday610