Xilinx/XRT #7720: Consult XRT for power on startup sequence logic

crizy, 2023-09-25:
I have looked at the source code of the xrt driver, and its power-up sequence is roughly as follows:
(1)mgmtpf/userpf  initialization,  create/probe sub device separately
    when userpf‘s all sub device are successfully probe and initialize,userpf launch the mailbox server
    when mgmtpf's all sub device are successfully probe and initialize(Including downloading boot firmware and Load 
    microblaze),launch the mailbox server，send mgmtpf online notification mailbox request 12 to usertpf
(2) after usertpf received mgmt driver online，userpf send mailbox request 12 with XCL_SUBDEV(6) sub request 
(3) after mgmg received XCL_SUBDEV(6) sub request，mgmtpf reply response
(4) after usertpf received response, userpf  call xocl_mb_connect(xdev)，userpf  get xocl_mb_connect status

But how did the other processes start afterwards ? for example, XCL_SENSOR/XCL_ICAP sub request

crizy, 2023-09-26:
I checked the code related to the XCL_ICAP sub request, and it seems to be a loopback between icap and mgmt?icap sends XCL_ICAP sub request, and the mgmt calls the icap to send XCL_ICAP sub request after receiving XCL_ICAP sub request？

uday610, 2023-10-05:
@houlz0507