Xilinx/XRT #7806: XRT not compiling on Ubuntu 23.10

keryell, 2023-11-17:
I did not try to compile XRT for 2 months and it looks like in the meantime I upgrade it from Ubuntu 23.04 to 23.10 and it does not compile anymore with GCC 13. It looks like some basic header include errors and declarations.
It would be nice to use the public GitHub action at least to check that it does just compile on a various landscape of compiler and distribution versions.
```
-<%>- ./build.sh                                                                                                                                                                                                     
Updating Git XRT submodule, use -noinit option to avoid updating
cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DXRT_ENABLE_WERROR=1 -DCMAKE_BUILD_TYPE=Debug ../../src
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found UnixCommands: /usr/bin/bash  
-- Host system processor is x86_64
-- Target system processor is x86_64
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1") 
-- Checking for module 'libdrm'
--   Found libdrm, version 2.4.115
-- Looking for DRM - found at /usr 2.4.115
-- Checking for module 'OpenCL'
--   Found OpenCL, version 3.0
-- Looking for OPENCL - found at /usr 3.0 /usr/include
-- Found Git: /usr/bin/git (found version "2.40.1") 
-- Looking for GIT - found at /usr/bin/git
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0") found components: system filesystem program_options 
-- Found Curses: /usr/lib/x86_64-linux-gnu/libcurses.so  
-- XRT EA eula files  /home/rkeryell/Xilinx/Projects/Vitis/XRT/src/../LICENSE
-- Platform/Linux (ubuntu) (Kernel 6.5.0-10-generic)
-- Compiler: /usr/bin/c++ /usr/bin/cc
-- add_subdirectory(runtime_src)
-- add_subdirectory(xdp)
-- add_subdirectory(tools/xclbinutil)
-- Found OpenSSL: /usr/lib/x86_64-linux-gnu/libcrypto.so (found version "3.0.10")  
-- RapidJSON found. Headers: /usr/include
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found GTest: /usr/lib/x86_64-linux-gnu/cmake/GTest/GTestConfig.cmake (found version "1.13.0")  
-- add_subdirectory(xocl)
-- add_subdirectory(xrt)
-- XRT xrt++ header files
-- xrt++.hpp
-- xrtexec.hpp
-- add_subdirectory(ert)
-- MicroBlaze toolchain found, preparing ERT build
-- Downloading sched_bsp.tar.bz2 from https://github.com/Xilinx/ERT-BSP/raw/main/BSPs/sched_bsp.tar.bz2
-- Downloading sched_bsp_u50.tar.bz2 from https://github.com/Xilinx/ERT-BSP/raw/main/BSPs/sched_bsp_u50.tar.bz2
-- Downloading sched_bsp_v20.tar.bz2 from https://github.com/Xilinx/ERT-BSP/raw/main/BSPs/sched_bsp_v20.tar.bz2
-- Downloading sched_bsp_v30.tar.bz2 from https://github.com/Xilinx/ERT-BSP/raw/main/BSPs/sched_bsp_v30.tar.bz2
-- add_subdirectory(tools/scripts)
-- add_subdirectory(core)
-- XRT deprecated header files
-- xrt.h
-- xcl_app_debug.h
-- xcl_axi_checker_codes.h
-- xclperf.h
-- add_subdirectory(common)
-- enabling api aie build
-- add_subdirectory(pcie)
-- add_subdirectory(common)
-- add_subdirectory(tools)
-- add_subdirectory(linux)
-- add_subdirectory(driver)
-- XRT header files
[... 356 lines cut ...]
[ 23%] Building CXX object runtime_src/xocl/CMakeFiles/xocl.dir/api/xlnx/clGetMemObjDeviceAddress.cpp.o
[ 23%] Building CXX object runtime_src/xocl/CMakeFiles/xocl.dir/api/xlnx/xclGetComputeUnitInfo.cpp.o
[ 23%] Building CXX object runtime_src/xocl/CMakeFiles/xocl.dir/api/xlnx/xclGetMemObjectFd.cpp.o
[ 23%] Building CXX object runtime_src/xocl/CMakeFiles/xocl.dir/api/xlnx/xclGetMemObjectFromFd.cpp.o
[ 23%] Building CXX object runtime_src/xocl/CMakeFiles/xocl.dir/api/xlnx/xclGetXrtDevice.cpp.o
[ 23%] Building CXX object runtime_src/xocl/CMakeFiles/xocl.dir/core/command_queue.cpp.o
[ 24%] Building CXX object runtime_src/xocl/CMakeFiles/xocl.dir/core/compute_unit.cpp.o
[ 24%] Building CXX object runtime_src/xocl/CMakeFiles/xocl.dir/core/context.cpp.o
[ 24%] Building CXX object runtime_src/xocl/CMakeFiles/xocl.dir/core/debug.cpp.o
[ 24%] Building CXX object runtime_src/xocl/CMakeFiles/xocl.dir/core/device.cpp.o
[ 24%] Building CXX object runtime_src/xocl/CMakeFiles/xocl.dir/core/event.cpp.o
[ 24%] Building CXX object runtime_src/xocl/CMakeFiles/xocl.dir/core/execution_context.cpp.o
[ 24%] Building CXX object runtime_src/xocl/CMakeFiles/xocl.dir/core/kernel.cpp.o
[ 24%] Building CXX object runtime_src/xocl/CMakeFiles/xocl.dir/core/memory.cpp.o
[ 25%] Building CXX object runtime_src/xocl/CMakeFiles/xocl.dir/core/platform.cpp.o
[ 25%] Building CXX object runtime_src/xocl/CMakeFiles/xocl.dir/core/program.cpp.o
[ 25%] Building CXX object runtime_src/xocl/CMakeFiles/xocl.dir/core/sampler.cpp.o
[ 25%] Building CXX object runtime_src/xocl/CMakeFiles/xocl.dir/xclbin/xclbin.cpp.o
[ 25%] Built target xocl
make: *** [Makefile:166: all] Error 2

real	1m13.319s
user	11m22.755s
sys	1m5.708s
```

dguenzel, 2023-11-20:
Hi,

you can find the officially supported operating systems in UG1451. Here is the document for the newest XRT 2023.2 (not yet officially released?): https://docs.xilinx.com/r/en-US/ug1451-xrt-release-notes

As you can see, only LTS releases of Ubuntu are supported and it was more or less a lucky coincidence that Ubuntu 23.04 worked for you. I'm just another user like you and not affiliated with Xilinx, but maybe this clarifies things a little.

Best regards.

keryell, 2023-11-20:
Hi!
Thank you for pointing me to the release notes.
Yes, I know it is not supported, but I just™ need it to work™, not to be supported™. ;-)
It looks more like an incorrect source file exposed by a modern compiler rather than by a newer OS version per se.
So this is basic code hygiene for an open source project.
I am trying to help an internal AMD research project involving FPGA, AIE and GPU, so the more planets are aligned, better it has chances to work.

keryell, 2024-03-11:
Thanks for fixing this.
Now it is time to figure out the failures with Linux 6.8...

closed 2024-03-11 by keryell