Xilinx/XRT #7810: Request to remove RTTI requirement

nirvedhmeshram, 2023-11-20:
Currently some parts of the XRT C++ API require use of RTTI which down stream projects may have disabled. To see the issue you may build this snippet.

https://gist.github.com/nirvedhmeshram/eb051397606439aa50016780ead8cf00

It will not work with the ollowing compile command
```
clang++ -fno-rtti test.cpp -o test.exe -I/opt/xilinx/xrt/include -L/opt/xilinx/xrt/lib -lxrt_coreutil
```
if you remove `fno-rtti` flag then it works but down stream projects may choose to not use rtti in which case we have problems.

keryell, 2023-11-20:
I guess you need to compile XRT without RTTI at the first place to have it to work.

stsoe, 2023-12-04:
XRT is built with rtti enabled which is the default in modern C++.   I don't understand the aversion to building an application with rtti enabled, but regardless you don't have to, but you just can't use the `info` API which requires rtti.  No other APIs rely on rtti being enabled.