Xilinx/XRT #7909: Build failure(s) with pybind11

spaceotter, 2024-01-25:
First, the xrtdeps.sh script install pybind11 with pip as sudo, which is not recommended, the xrt build script also then fails to detect it. I installed pybind11 with a virtualenv and this was found by cmake.

Next, there is a build error:
```
In file included from /usr/include/pybind11/cast.h:16,
                 from /usr/include/pybind11/attr.h:13,
                 from /usr/include/pybind11/pybind11.h:13,
                 from /home/eric/src/xdna-driver/xrt/src/python/pybind11/src/pyxrt.cpp:20:                              /usr/include/pybind11/detail/type_caster_base.h: In function ‘std::string pybind11::detail::error_string()’:            /usr/include/pybind11/detail/type_caster_base.h:482:26: error: invalid use of incomplete type ‘PyFrameObject’ {aka ‘struct _frame’}
  482 |             frame = frame->f_back;                                                                                    |                          ^~
```
I'm guessing it's related to this: https://docs.python.org/3/whatsnew/3.11.html#pyframeobject-3-11-hiding
Even though I'm using python 3.10, maybe pybind updated for 3.11 and this affected 3.10?

spaceotter, 2024-01-26:
The problems seem to originate from having python3.11 and/or pybind11-dev installed. In this case, for whatever reason the XRT build scripts always pick up the system python 3.11 even you activate a virtualenv. The xrtdeps.sh script installs the pybind11 dependency with the default python, which in ubuntu 22 is 3.10. And the pybind11-dev package is old and not compatible with 3.11 due to previously linked reason.

Quick fix is to use `sudo python3.11 -m pip install pybind11` - but that's not a great solution. I'd rather like to know why the XRT build script completely ignores the user environment...

ManojTakasi, 2024-08-07:
Hi @spaceotter , we have added a check in our CMakeLists.txt to determine if user is in a virtual environment. Based on that, we are setting the python3 exectable accordingly. With this PR i hpoe your issue is resolved. If you think it remains unresolved, please free to reopen this issue.

Thank you.

closed 2024-08-07 by ManojTakasi