Xilinx/XRT #7911: Potential buffer overflow in error report

stellaraccident, 2024-01-26:
Getting pages of these for different source files:

```
[481/496] Building CXX object src/runtime_src/core/pcie/linux/CMakeFiles/core_pcielinux_objects.dir/device_linux.cpp.o
In file included from /home/stella/src/iree-amd-aie/third_party/XRT/src/runtime_src/core/pcie/linux/device_linux.cpp:8:
/home/stella/src/iree-amd-aie/third_party/XRT/src/runtime_src/core/common/query_requests.h:942:72: warning: adding 'const xrt_core::query::device_status::result_type' (aka 'const unsigned int') to a string does not append to the string [-Wstring-plus-int]
        throw xrt_core::system_error(EINVAL, "Invalid device status: " + status);
                                             ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/home/stella/src/iree-amd-aie/third_party/XRT/src/runtime_src/core/common/query_requests.h:942:72: note: use array indexing to silence this warning
        throw xrt_core::system_error(EINVAL, "Invalid device status: " + status);
                                                                       ^
                                             &                         [       ]
/home/stella/src/iree-amd-aie/third_party/XRT/src/runtime_src/core/common/query_requests.h:3418:77: warning: adding 'const xrt_core::query::performance_mode::result_type' (aka 'const unsigned int') to a string does not append to the string [-Wstring-plus-int]
        throw xrt_core::system_error(EINVAL, "Invalid performance status: " + status);
                                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/home/stella/src/iree-amd-aie/third_party/XRT/src/runtime_src/core/common/query_requests.h:3418:77: note: use array indexing to silence this warning
        throw xrt_core::system_error(EINVAL, "Invalid performance status: " + status);
                                                                            ^
                                             &                              [       ]
2 warnings generated.
```

I think this is a legit error and should be fixed.

I'm getting something like it on recent gcc and clang. The clang version is 14.0.0. Please don't just silence the warning -- this is a legit bug.

dbenusov-xilinx, 2024-01-26:
@stellaraccident Thanks for pointing this out! I will update the code to use the parse_status function.

dbenusov-xilinx, 2024-01-29:
#7916

dbenusov-xilinx, 2024-01-31:
@stellaraccident Please let me know if the warnings stop with the update in XRT.

closed 2024-02-06 by dbenusov-xilinx