Xilinx/XRT #7912: XRT build drops a `kernel-doc` directory in the PWD

stellaraccident, 2024-01-27:
When I invoke the build from an arbitrary location, it drops a `kernel-doc` directory in the current PWD. Recommend finding this and scoping it to the build dir (or something).

ManojTakasi, 2024-02-06:
Hi @stellaraccident, I have tried building XRT from arbitrary locations. However, I'm unable to reproduce the issue. Could you please share the steps you took to encounter this issue.

stellaraccident, 2024-02-06:
Sure. I noticed it after some iteration in this: https://github.com/nod-ai/iree-amd-aie?tab=readme-ov-file#building-xrt

We ended up accidentally checking the file in and adding it to gitignore: https://github.com/nod-ai/iree-amd-aie/commit/a42ddaca69c1068f242c08cd204232f3078b2040

ManojTakasi, 2024-03-19:
PR has been merged into master