Xilinx/XRT #8093: XRT 2022.1/2022.2 build.sh fails on xocl using ubuntu 22.04 kernel ver 6.5

dirksavage88, 2024-04-22:
Similar output to https://github.com/Xilinx/XRT/issues/7806. I set the XILINX_VITIS environment variable to the Vitis install directory.

[debug.txt](https://github.com/Xilinx/XRT/files/15056230/debug.txt)

keryell, 2024-04-23:
Please use the latest XRT.

dirksavage88, 2024-04-24:
Same error using the latest XRT (master) AND I ran xrtdeps.sh
libudev is showing installed:

`Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libudev-dev is already the newest version (249.11-0ubuntu3.12).
0 upgraded, 0 newly installed, 0 to remove and 57 not upgraded.
`
Path variable:

`echo $PATH
/home/papadrew/tools/Xilinx/Vitis_HLS/2022.1/bin:/home/papadrew/tools/Xilinx/Vivado/2022.1/bin:/home/papadrew/tools/Xilinx/Vitis/2022.1/bin:/home/papadrew/tools/Xilinx/Vitis/2022.1/gnu/microblaze/lin/bin:/home/papadrew/tools/Xilinx/Vitis/2022.1/gnu/arm/lin/bin:/home/papadrew/tools/Xilinx/Vitis/2022.1/gnu/microblaze/linux_toolchain/lin64_le/bin:/home/papadrew/tools/Xilinx/Vitis/2022.1/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/home/papadrew/tools/Xilinx/Vitis/2022.1/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/home/papadrew/tools/Xilinx/Vitis/2022.1/gnu/aarch64/lin/aarch64-linux/bin:/home/papadrew/tools/Xilinx/Vitis/2022.1/gnu/aarch64/lin/aarch64-none/bin:/home/papadrew/tools/Xilinx/Vitis/2022.1/gnu/armr5/lin/gcc-arm-none-eabi/bin:/home/papadrew/tools/Xilinx/Vitis/2022.1/aietools/bin:/opt/gcc-arm-none-eabi-9-2020-q2-update/bin:/opt/ros/humble/bin:/home/papadrew/.sdkman/candidates/gradle/current/bin:/opt/gcc-arm-none-eabi-7-2017-q4-major/bin:/opt/gcc-arm-none-eabi-9-2020-q2-update/bin:/home/papadrew/.local/bin:/home/papadrew/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin`

kernel:

`6.5.0-27-generic #28~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Mar 15 10:51:06 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
`

keryell, 2024-04-24:
To be clearer, try with a clean environment, with just standard minimal Linux `PATH`.

dirksavage88, 2024-05-26:
Clean PATH and still not finding libudev:
```
make package
Consolidate compiler generated dependencies of target core_common_xdp_profile_objects
[  0%] Built target core_common_xdp_profile_objects
Consolidate compiler generated dependencies of target core_common_library_objects
[  3%] Built target core_common_library_objects
Consolidate compiler generated dependencies of target core_common_api_library_objects
[  6%] Built target core_common_api_library_objects
[  6%] Built target xrt_coreutil
Consolidate compiler generated dependencies of target xocl
[ 24%] Built target xocl
[ 24%] Building CXX object runtime_src/xrt/CMakeFiles/xrt++.dir/device/device.cpp.o
[ 24%] Building CXX object runtime_src/xrt/CMakeFiles/xrt++.dir/device/hal.cpp.o
[ 24%] Building CXX object runtime_src/xrt/CMakeFiles/xrt++.dir/device/hal2.cpp.o
[ 24%] Building CXX object runtime_src/xrt/CMakeFiles/xrt++.dir/xrt++/xrtexec.cpp.o
[ 24%] Linking CXX shared library libxrt++.so
[ 24%] Built target xrt++
[ 24%] Linking CXX shared library libxilinxopencl.so
[ 24%] Built target xilinxopencl
[ 24%] Building CXX object runtime_src/xdp/appdebug/CMakeFiles/xdp_appdebug_plugin.dir/appdebug.cpp.o
In file included from /usr/include/ocl_icd.h:46,
                 from /home/papadrew/tools/XRT/src/runtime_src/xocl/api/icd/ocl_icd_bindings.h:28,
                 from /home/papadrew/tools/XRT/src/runtime_src/xocl/core/object.h:7,
                 from /home/papadrew/tools/XRT/src/runtime_src/xdp/appdebug/appdebug.h:26,
                 from /home/papadrew/tools/XRT/src/runtime_src/xdp/appdebug/appdebug.cpp:27:
/usr/include/CL/cl_gl_ext.h:18:112: note: ‘#pragma message: All OpenGL-related extensions have been moved into cl_gl.h.  Please include cl_gl.h directly.’
   18 | OpenGL-related extensions have been moved into cl_gl.h.  Please include cl_gl.h directly.")
      |                                                                                           ^

[ 24%] Building CXX object runtime_src/xdp/appdebug/CMakeFiles/xdp_appdebug_plugin.dir/appdebug_plugin.cpp.o
In file included from /usr/include/ocl_icd.h:46,
                 from /home/papadrew/tools/XRT/src/runtime_src/xocl/api/icd/ocl_icd_bindings.h:28,
                 from /home/papadrew/tools/XRT/src/runtime_src/xocl/core/object.h:7,
                 from /home/papadrew/tools/XRT/src/runtime_src/xocl/core/platform.h:20,
                 from /home/papadrew/tools/XRT/src/runtime_src/xdp/appdebug/appdebug_plugin.h:20,
                 from /home/papadrew/tools/XRT/src/runtime_src/xdp/appdebug/appdebug_plugin.cpp:17:
/usr/include/CL/cl_gl_ext.h:18:112: note: ‘#pragma message: All OpenGL-related extensions have been moved into cl_gl.h.  Please include cl_gl.h directly.’
   18 | OpenGL-related extensions have been moved into cl_gl.h.  Please include cl_gl.h directly.")
      |                                                                                           ^

[ 24%] Building CXX object runtime_src/xdp/appdebug/CMakeFiles/xdp_appdebug_plugin.dir/appdebugmanager.cpp.o
In file included from /usr/include/ocl_icd.h:46,
                 from /home/papadrew/tools/XRT/src/runtime_src/xocl/api/icd/ocl_icd_bindings.h:28,
                 from /home/papadrew/tools/XRT/src/runtime_src/xocl/core/object.h:7,
                 from /home/papadrew/tools/XRT/src/runtime_src/xocl/core/platform.h:20,
                 from /home/papadrew/tools/XRT/src/runtime_src/xdp/appdebug/appdebugmanager.h:20,
                 from /home/papadrew/tools/XRT/src/runtime_src/xdp/appdebug/appdebugmanager.cpp:17:
/usr/include/CL/cl_gl_ext.h:18:112: note: ‘#pragma message: All OpenGL-related extensions have been moved into cl_gl.h.  Please include cl_gl.h directly.’
   18 | OpenGL-related extensions have been moved into cl_gl.h.  Please include cl_gl.h directly.")
      |                                                                                           ^

[ 25%] Linking CXX shared module libxdp_appdebug_plugin.so
[ 25%] Built target xdp_appdebug_plugin
[ 25%] Building CXX object runtime_src/xdp/debug/CMakeFiles/xdp_debug_plugin.dir/debug_plugin.cpp.o
[ 25%] Building CXX object runtime_src/xdp/debug/CMakeFiles/xdp_debug_plugin.dir/kernel_debug_manager.cpp.o
[ 25%] Linking CXX shared module libxdp_debug_plugin.so
[ 25%] Built target xdp_debug_plugin
[ 25%] Building CXX object runtime_src/xdp/profile/CMakeFiles/xdp_core.dir/database/database.cpp.o
[ 25%] Building CXX object runtime_src/xdp/profile/CMakeFiles/xdp_core.dir/database/dynamic_event_database.cpp.o
[ 25%] Building CXX object runtime_src/xdp/profile/CMakeFiles/xdp_core.dir/database/dynamic_info/aie_db.cpp.o
[ 25%] Building CXX object runtime_src/xdp/profile/CMakeFiles/xdp_core.dir/database/dynamic_info/dependency_manager.cpp.o
[ 25%] Building CXX object runtime_src/xdp/profile/CMakeFiles/xdp_core.dir/database/dynamic_info/host_db.cpp.o
[ 25%] Building CXX object runtime_src/xdp/profile/CMakeFiles/xdp_core.dir/database/dynamic_info/pl_db.cpp.o
[ 25%] Building CXX object runtime_src/xdp/profile/CMakeFiles/xdp_core.dir/database/dynamic_info/string_table.cpp.o
[ 26%] Building CXX object runtime_src/xdp/profile/CMakeFiles/xdp_core.dir/database/events/creator/aie_trace_data_logger.cpp.o
[ 26%] Building CXX object runtime_src/xdp/profile/CMakeFiles/xdp_core.dir/database/events/device_events.cpp.o
[ 26%] Building CXX object runtime_src/xdp/profile/CMakeFiles/xdp_core.dir/database/events/hal_api_calls.cpp.o
[... cut ...]
[ 70%] Built target xrt_core_static
Consolidate compiler generated dependencies of target mpd
[ 70%] Linking CXX executable mpd
/usr/bin/ld: cannot find -ludev: No such file or directory
collect2: error: ld returned 1 exit status
make[2]: *** [runtime_src/core/pcie/tools/cloud-daemon/CMakeFiles/mpd.dir/build.make:148: runtime_src/core/pcie/tools/cloud-daemon/mpd] Error 1
make[1]: *** [CMakeFiles/Makefile2:2591: runtime_src/core/pcie/tools/cloud-daemon/CMakeFiles/mpd.dir/all] Error 2
make: *** [Makefile:166: all] Error 2

```

dirksavage88, 2024-11-25:
> Same exact situation. Did you ever find a solution @dirksavage88 ?

Nope. I had the whole toolchain originally working on ubuntu 20.04, but as soon as I upgraded to 22.04 and updated the toolchain I ran into this issue, which is frustrating. I'm considering going back to 20.04 and trying again.