Xilinx/XRT #8124: Coexistence of OpenCL and XRT APIs and long-term support

qberthet, 2024-04-30:
In the context of the ATLAS experiment at CERN, we are exploring the potential utilization of AMD FPGA accelerators in a forthcoming computing farm. Presently, we have multiple teams engaged in independent work on kernel algorithms, with varying preferences regarding the use of OpenCL versus the Native XRT API. We aim to initiate the integration of these efforts into the existing software framework.

Question 1: Is it feasible to employ both the Native XRT and OpenCL APIs within the same software solution? In other words, can a device programmed with the Native XRT API be utilized alongside the OpenCL API (Without reloading the XCLBIN with OpenCL)? If so, it would be great to have some pointers on how to do achieve this.

Question 2: Does AMD provide an official recommendation regarding the preferred API choice? We seek guidance on factors such as features, performance, and long-term support, as the decision between the two APIs is currently challenging based on the available online information.

stsoe, 2024-07-10:
@qberthet 

Q1: OpenCL is on top of XRT native APIs, the interoperability is from OpenCL to XRT APIs, not the other way around.  I believe the OpenCL implementation creates a lot of data from the clAPIs, which is not something that can be created from XRT APIs.    You can go from OpenCL -> XRT, but only for the most basisc.  There is a header file include/CL/cl2xrt.hpp which maps some OpenCL objects to XRT.

Q2: XRT is recommended.  OpenCL is overhead.  All new features are added to XRT.  For example, XRT is the only APIs supported for NPU.

qberthet, 2024-07-11:
Thanks for the answer @stsoe. Yes, I have since figured out that it was possible to get handles on underlying XRT objects from the OpenCL API using include/CL/cl2xrt.hpp. I successfully ran some tests with code using both APIs simultaneously (I know this sounds like a waste of effort, but it was necessary for us to check this on our side). I forgot to report that here, sorry about that.

About XRT/OpenCL: that was also my understanding, thanks for the confirmation.

closed 2024-07-11 by qberthet