Xilinx/XRT #8343: 403  Forbidden during install attempt using apt

daverin, 2024-08-10:
My docker builds worked a few days ago but are now all failing during xrt package install on ubuntu 20.04

```
Err:1 https://packages.xilinx.com/artifactory/debian-packages focal/main amd64 xrt amd64 2.16.204
  403  Forbidden [IP: 52.85.42.115 443]
E: Failed to fetch https://dwgrutst0mjb3.cloudfront.net/filestore/57/57810aac976969bbe2a2d017671cfb92b89feddb?response-content-type=application/x-debian-package&x-jf-traceId=3fcac6f00441ff4d&X-Artifactory-repositoryKey=debian-packages-cache&X-Artifactory-projectKey=default&X-Artifactory-artifactPath=pool/xrt_202320.2.16.204_20.04-amd64-xrt.deb&X-Artifactory-username=anonymous&Expires=1723263472&Signature=J~mXN32oCIaN-aoXUtMPPn0V-s6OksAKwFSBFYTGC~4P8GiuEYPsQ9ACl8gpQKY4x-H0k2pU1JL3P~LYdF2B7VtKFXyPxHKRHrAhpHgjP8dbllw~QEC0vRm40II2kUVzNaT52Y7obapub83mKsfMPYbT1sFbYZIa-rraSzg3WdssjOk2rJmCM~2RqdivqOnzoEDjFs2EAfqs2cP7TC6O7ZMu6o5QdsShgyyzDEFf~zu~cxIhG3d~tkzU5APGajb9BYhIJUTJJtHn4jR4RXHpd~va46rg-PtF17EcAlvMik5ke6CnoE1PPgUkSWk~qOXd16XkmrVJFNxj1kujVddBEA__&Key-Pair-Id=K1N32JWIQUWNE3  403  Forbidden [IP: 52.85.42.115 443]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

ybonnel, 2024-08-16:
@daverin I've exactly the same problem, did you find a solution?

bhargav-sudo, 2024-09-01:
Facing this same issue. Any solution to this .. 😢

daverin, 2024-09-03:
Hi @ybonnel and @bhargav-sudo . Checkout https://github.com/Xilinx/video-sdk/issues/95

Haven't tried it yet, but looks like the solution.

bhargav-sudo, 2024-09-03:
Thanks @daverin . I tried this it worked 😄