Xilinx/XRT #8638: Build xrt-smi Utility on Windows

a-carrara, 2024-12-03:
**XRT Version:** master, commit: `3ee37cfefc32047fea8d334164844e900ad8e7ef`  
**Platform:** Windows  

Hello there,  

I am trying to compile the `xrt-smi` tool on Windows, but I can't find any step-by-step guide on how to do this.  

For context, I am exploring ways to diagnose the health of NPU devices. I discovered this tool in the [NPU Management Interface — Ryzen AI Software 1.3 documentation](https://ryzenai.docs.amd.com/en/latest/xrt_smi.html). After installing the Ryzen AI tool, I was able to find and use the `xrt-smi` utility as expected.  

However, I want to understand more deeply what happens during these tests and leverage the same API that the utility uses in my own application. When I attempted to compile the tool myself, I encountered an issue.  

To be precise, I was able to compile the tool by using the `build/build22.bat` file, but the compiled version behaves differently from the one installed on my machine. Specifically, it does not detect my NPU device. I wonder if I need to configure a specific environment variable or define certain macros to achieve the same result as the installed version.  

Could you provide guidance or documentation on how to properly build the `xrt-smi` tool on Windows? Also, if this is not the correct place to ask this type of question, please let me know.  

Thank you in advance!

stsoe, 2024-12-04:
@a-carrara.   Make sure the hash of https://github.com/Xilinx/XRT you pull to build `xrt-smi` corresponds to the hash listed by running `xrt-smi examine` on your system.   `xrt-smi` must be in sync with driver and libraries installed on your system.

Let me know if you continue to run into problems.

a-carrara, 2024-12-04:
@stsoe Alright, I'll give that a try and get back to you with the results.

a-carrara, 2024-12-09:
@stsoe It Works! Also, to run with my versions of xrt_core.dll and xrt_coreutil.dll, I had to define the macro XRT_WINDOWS_HAS_WDK before compiling everything.

stsoe, 2024-12-09:
@a-carrara  Good to hear it works for you.  

Your own xrt_core.dll is not at all used by NPU, it is an Alveo library.   For NPU, libxrt_core.dll is so far not even open source and the binary is loaded from the Windows driver store.  

When you install the driver for NPU, xrt_coreutil.dll is installed in windows/system32/ folder and libxrt_core.dll in the driver store.  xrt-smi (and any user application) uses xrt_coreutil.dll, which in turn loads libxrt_corel.dll at runtime.

When I advised you to make sure XRT was at the git hash reported by xrt-smi, it was because xrt-smi indirectly loads xrt_core from the driver store and the version of xrt_core must match the version of xrt_coreutil.

a-carrara, 2024-12-09:
@stsoe I see, thanks for the clarification and all the support, the information provided was very helpful.  For now I have no more questions.