Xilinx/XRT #8678: xrt-smi does not work in Ryzen1.3 after installation following 1.3 documentation

MLValidationDebug, 2025-01-02:
xrt-smi does not return anything, 
(ryzen-ai-1.3.0) PS U:\> C:\Windows\System32\AMD\xrt-smi.exe
(ryzen-ai-1.3.0) PS U:\>

I suspect a dll link missing or error, and I use Dependency Walker to analyse xrt-smi exe, it does report some dll missing, but I don't know if I can trust it or not. 
I've try to follow the installation precedure for ryzen1.3, and npu driver are updated to 237 the latest one. I've run some demos/tutorials following the docs, so settings/configuration/driver etc. are working fine.
https://ryzenai.docs.amd.com/en/latest/xrt_smi.html

Other system infor as below:
Processor	AMD Ryzen 7 7840U w/ Radeon  780M Graphics        3.30 GHz
Installed RAM	32.0 GB (31.3 GB usable)
Edition	Windows 11 Enterprise
Version	22H2
OS build	22621.4460
Experience	Windows Feature Experience Pack 1000.22700.1047.0

rainerhosch, 2025-04-30:
hi ther, are u have installed the xrt-smi utility requires Microsoft Visual C++ Redistributable?
[Microsoft Visual C++ Redistributable](https://learn.microsoft.com/en-us/cpp/windows/latest-supported-vc-redist?view=msvc-170) (version 2015 to 2022)

MLValidationDebug, 2025-04-30:
> Caution: This message originated from an External Source. Use proper caution when opening attachments, clicking links, or responding.

[AMD Official Use Only - AMD Internal Distribution Only]

Thanks for reaching out, I believe I did installed VC distribution 2022, I follow the documentation quite strictly.

The problem is I don't know where to look at, since no output from the command at all. I unfortunately don't have the environment anymore.