Xilinx/XRT #8790: Build fails for `./build.sh -npu -opt` on Ubuntu 24.10

jgmelber, 2025-02-25:
This is building for use with the `xdna-driver` @ [8b4eefc](https://github.com/amd/xdna-driver/commit/8b4eefc403b89e5d44d771c75a6bef5d55087bfd) and `XRT` @f02de266f785111094b02d32a9012bd0da3abd58
```
-- Decoding ascii file /home/jmelber/xdna-driver/xrt/src/runtime_src/core/common/aiebu/test/aie2-ctrlcode/basic/ml_txn.b64 to binary file ml_txn.bin
-- Decoding ascii file /home/jmelber/xdna-driver/xrt/src/runtime_src/core/common/aiebu/test/aie2-ctrlcode/basic/ctrl_pkt0.b64 to binary file ctrl_pkt0.bin
/home/jmelber/xdna-driver/xrt/src/runtime_src/core/common/aiebu/lib/aie-rt/driver/src/common/xaie_helper.c: In function ‘XAie_Txn_MergeSync’:
/home/jmelber/xdna-driver/xrt/src/runtime_src/core/common/aiebu/lib/aie-rt/driver/src/common/xaie_helper.c:2861:57: error: ‘calloc’ sizes specified with ‘sizeof’ in the earlier argument and not in the later argument [-Werror=calloc-transposed-args]
 2861 |                 u32* tctDataBuff = (u32 *)calloc(sizeof(tct_op_t), 1);
      |                                                         ^~~~~~~~
/home/jmelber/xdna-driver/xrt/src/runtime_src/core/common/aiebu/lib/aie-rt/driver/src/common/xaie_helper.c:2861:57: note: earlier argument should specify number of elements, later size of each element
/home/jmelber/xdna-driver/xrt/src/runtime_src/core/common/aiebu/lib/aie-rt/driver/src/common/xaie_helper.c: In function ‘XAie_Txn_DdrAddressPatch’:
/home/jmelber/xdna-driver/xrt/src/runtime_src/core/common/aiebu/lib/aie-rt/driver/src/common/xaie_helper.c:2917:73: error: ‘calloc’ sizes specified with ‘sizeof’ in the earlier argument and not in the later argument [-Werror=calloc-transposed-args]
 2917 |                 patch_op_t* patchDataBuff = (patch_op_t *)calloc(sizeof(patch_op_t), 1);
      |                                                                         ^~~~~~~~~~
/home/jmelber/xdna-driver/xrt/src/runtime_src/core/common/aiebu/lib/aie-rt/driver/src/common/xaie_helper.c:2917:73: note: earlier argument should specify number of elements, later size of each element
```

jgmelber, 2025-02-25:
https://github.com/amd/xdna-driver/issues/391

stsoe, 2025-02-26:
See #8764.

jgmelber, 2025-02-26:
> See [#8764](https://github.com/Xilinx/XRT/issues/8764).

Thanks. I will close this and leave the fix to #8764.

closed 2025-02-26 by jgmelber