Xilinx/XRT #8908: Access registers (read_register / write_register) without loading (or having) xclbin file

teqdruid, 2025-04-24:
Is it possible to read and write registers without loading an xclbin file?

I've figured out how to get the UUID out of the device, but when I construct a `xrt::ip` giving it the device, the discovered UUID, and the known kernel name I get `xclbin id mismatch: Invalid argument` exception.

Use case: I'm writing an infrastructure which uses a standard kernel name, base register offset, and a standard set of registers for discovery. It provides a utility application which can connect to a device and discover various information about the image being run -- it should _not_ load the xclbin since that would defeat the purpose.

teqdruid, 2025-05-07:
Any updates on this? If this is not possible, is it feasible to add this feature?

matthew-humphrey, 2025-05-09:
I would really like to know the answer to this as well.